Azure/iot-edge-modbus #3: Executing Modbus Module Example Throws OpenSSL Error

WilliamBerryiii, 2016-12-21:
When the current version of the module is built and run against the `2016-11-18` branch of the azure-iot-gateway-sdk the following error is thrown indicating a failure to establish a connection to the IoT Hub:

```
Error: Time:Wed Dec 21 12:58:55 2016 File:/home/pi/code/azure-iot-gateway-sdk/deps/c-utility/src/tlsio_openssl.c Func:log_ERR_get_error Line:296 Failed allocating OpenSSL context.
Error: Time:Wed Dec 21 12:58:55 2016 File:/home/pi/code/azure-iot-gateway-sdk/deps/c-utility/src/tlsio_openssl.c Func:log_ERR_get_error Line:303   [0] error:140A90A1:lib(20):func(169):reason(161)
Error: Time:Wed Dec 21 12:58:55 2016 File:/home/pi/code/azure-iot-gateway-sdk/deps/iot-sdk/c/iothub_client/src/iothubtransport_amqp_common.c Func:establishConnection Line:627 Failed to open the connection with CBS.
Error: Time:Wed Dec 21 12:58:55 2016 File:/home/pi/code/azure-iot-gateway-sdk/deps/iot-sdk/c/iothub_client/src/iothubtransport_amqp_common.c Func:IoTHubTransport_AMQP_Common_DoWork Line:1349 AMQP transport failed to establish connection with service.
Error: Time:Wed Dec 21 12:58:55 2016 File:/home/pi/code/azure-iot-gateway-sdk/deps/iot-sdk/c/iothub_client/src/iothubtransportamqp_auth.c Func:authentication_reset Line:637 Failed to reset the authentication state (authentication status is invalid: 0)
Error: Time:Wed Dec 21 12:58:55 2016 File:/home/pi/code/azure-iot-gateway-sdk/deps/iot-sdk/c/iothub_client/src/iothubtransport_amqp_common.c Func:prepareDeviceForConnectionRetry Line:1071 Failed resetting the authenticatication state of device �	�r
```

The IoTHub module's transport was set to AMQP as follows:

```
{
      "name": "IotHub",
      "loader": {
        "name": "native",
        "entrypoint": {
          "module.path": "../../modules/iothub/libiothub.so"
        }
      },
      "args": {
        "IoTHubName": "********",
        "IoTHubSuffix": "*********",
        "Transport": "AMQP"
      }
    },```

The error is not thrown with a similar IoT Hub and Device configuration in the `simulated_device_cloud_upload` sample project in the base SDK repository.

606simps, 2017-01-16:
Hi,
Thanks for reporting this. Try to use `simulated_device_cloud_upload` sample program as your main program and load it with `modbus_lin.json`. This is just a work around and I will update the fix later.

606simps, 2017-01-16:
Now done.

closed 2017-01-16 by 606simps