Azure/iot-edge-modbus #6: Error when building project

sumongithub, 2017-02-28:
-- Configuring done
-- Generating done
-- Build files have been written to: E:/.../IoT/Azure/azure-iot-gateway-sdk/deps/nanomsg/build
-- Installing nanomsg. Please wait...
CMake Error at samples/modbus_sample/CMakeLists.txt:38 (copy_gateway_dll):
  Unknown CMake command "copy_gateway_dll".

-- Configuring incomplete, errors occurred!
See also "E:/.../IoT/Azure/azure-iot-gateway-sdk/build/CMakeFiles/CMakeOutput.log".
See also "E:/.../IoT/Azure/azure-iot-gateway-sdk/build/CMakeFiles/CMakeError.log".

606simps, 2017-03-01:
Hi,

What versions of Gateway SDK and Modbus module are you using respectively?
Please provide more details about your building environment so I could reproduce the issue.
Thanks.

sumongithub, 2017-03-02:
Hi,
Thank your for your input.
Initially it was old version, then updated with git PULL. Now cloned with latest Gateway SDK and build was successful. 

Thanks

closed 2017-03-02 by sumongithub