Azure/iot-edge-modbus #7: Connecting to multiple modbus devices

janfl, 2017-04-25:
I need the same gateway to read from multiple ModBus enabled devices.

How can this be achieved with you module, i can't seam to get it working?

606simps, 2017-05-04:
Yes you can. Configure the gateway as below.
```json
"args": [
        {
          "serverConnectionString": "<Device1>",
          ...
          ...
        },        
        {
          "serverConnectionString": "<Device2>",
          ...
          ...
        },        
        {
          "serverConnectionString": "<Device3>",
          ...
          ...
        }
]

janfl, 2017-05-04:
Thanks it working.
When i use hostname instead of ip in serverConnectionString i can't connect, doesn't the module support this?

606simps, 2017-05-05:
No, hostname is currently not supported. However, you should be able to implement it since all source codes are available. You are also very welcome to contribute your work to the repo once you are done.
Thanks!

closed 2017-05-05 by 606simps