Azure/iot-edge-modbus #8: Extend module

janfl, 2017-05-17:
Hi
I'm very new to c++ and i want to extend your module but i can't get the code to build.
How should i extend you module and test if it can build without having to copy the folders to gateway folder?

606simps, 2017-05-25:
Hi,
Gateway SDK has been released as a nugget package, you can try to build the module by importing the Gateway package. You may need to modify some codes though.

janfl, 2017-05-29:
Now the Gateway SDK support nuget, do you have plans of releasing the modbus module as a VS project that can be build from Visual studio?

606simps, 2017-06-20:
The Gateway SDK nuget package now only supports .Net. This Modbus module is written in pure C, hence this cannot be done until Gateway SDK release nuget package for native C. I will close the issue and feel free to reopen issue if you find anything else.

closed 2017-06-20 by 606simps