Azure/iot-edge-modbus #9: Module do not resume communication if fail

janfl, 2017-05-18:
When standard modbus module fails sending in the send_request_tcp method it does not resume communication until i restart the Gateway. The error is "Send failed".

janfl, 2017-05-22:
Is this module not supported any more?
 If supported what can i expect in regards to response on issues?

606simps, 2017-05-25:
Yes the module is still supported. Sorry for the delayed response. Could you elaborate more details about the issue and also attach your log? Thanks.

janfl, 2017-05-29:
I have configured multiple(3) connections to modbus enabled hardware.
Some time the module run for long time without problems other times it only run for short period of time.
It logs "Send failed" in the send_request_tcp method and "send request failed" within the while (request_operation) loop and this of result in "unable to send request to modbus server" failure
After it has failed the first time for one of the connected hardware it stop receiving data from any of the connected hardware.
Do you connect to each configured modbus device one time and reuse this connection each time you retrieve data from it? Maybe you missing some reconnect on lost connection functionality either on actual lost connection or failure on reading data from modbus device....

606simps, 2017-06-26:
Sorry for the wait. I've committed the fix, please pull the latest code and try it out.

Thanks,
Stephen

closed 2017-06-26 by 606simps