Azure/iot-edge-modbus #10: Socket handling

jild, 2017-05-19:
Hi, in your code you are checking for return of recv == _SOCKET_ERROR_ that is -1 but you also have to check for == 0
[image]

See manpages: "These calls return the number of bytes received, or -1 if an error occurred. The return value will be 0 when the peer has performed an orderly shutdown." from https://linux.die.net/man/2/recv

Also the socket is stream oriented and may transfer fewer bytes than requested for read or write operations. This is referred to as socket _short read_ for the recv function. _recv_size_ needs to be checked, also check if the number of bytes to send are less compared with the return value of _send_ (short write).

See: http://stackoverflow.com/questions/666601/what-is-the-correct-way-of-reading-from-a-tcp-socket-in-c-c

606simps, 2017-05-25:
Thanks for the feedback. Will get it fixed soon.

606simps, 2017-05-25:
Thanks again for your help. Code fixed. Please feel free to leave your comment or open issues if you find anything.

closed 2017-05-25 by 606simps