Azure/iot-edge-modbus #14: Request for varying read intervals across registers

kiranpradeep, 2017-11-19:
There could be need to read different registers at different frequencies. With the current setup( of single "Interval" spanning all registers/slaves), lib user is forced set up configuration with least common denominator of all read frequencies( and then ignore data as appropriate ).

606simps, 2017-11-20:
Yes, that's correct, you need to work around the current limitation. I will add this in the next release. Or you are very welcome to contribute your work on this. Thanks.

kiranpradeep, 2017-11-23:
@606simps Sorry. I am not familiar with C#. Thank you for considering this.

606simps, 2017-12-04:
Now it has two interval settings in the configuration.
"PublishInterval": "interval to push data to the cloud"
"PollingInterval": "interval to poll data from Modbus device"
Hope this helps.

kiranpradeep, 2017-12-18:
Sorry for late reply. My request was only for varying read intervals. In addition to your core module, we also have a custom module for logic like data type, data aggregation etc. 

Use case:  Different registers need to be acquired at different intervals and published at different intervals. Custom module developer might need to do a different kind of logic on the data collected. It could be average on data, last data point only etc. With the current setup, I am not sure how to achieve this. Can you leave the job of publish interval to the custom module developer? Or is there any reason why we need publishInterval in the core module itself.

606simps, 2017-12-18:
Hi,

Hope I get your issue right. You are asking for two things.

1. Different intervals to acquire data from Modbus devices. (already implemented)
2. Different intervals to publish acquired data to edgeHub (route to cloud or custom modules). (currently share one single variable)

The reason we need publishInterval in the Modbus module is because we don't want to publish data as soon as data are collected from Modbus devices. We keep all data temporarily in a local buffer and publish them every publishInterval. If your custom module is the destination of Modbus data, you can still manage your own publishInterval of the custom module. Furthermore, we keep timestamp field for every register in the ModbusOutMessage structure, so you should still be able to do average of data and last data point etc.
Let me know if there are any other concerns.

Thanks.

closed 2017-12-26 by kiranpradeep