Azure/iot-edge-modbus #17: Recommendation - Data types

thegunshow, 2017-12-09:
Hi

This is not a problem but a recommendation, the data could be converted to the correct data type in the same module. It would be useful to have different data types e.g. unit16, int32, float32, etc.
This information could be incorporated into the device twin for the module when you build each modbus mapping.

Thanks
Brendan

606simps, 2017-12-14:
Hi Brendan,

I agree that it would be helpful to have the data converted to the correct data types. For now, this can be done in a separate custom module. We will consider integrate this function in the module.

Thanks,
Stephen

thegunshow, 2017-12-14:
Hi Stephen

If I add a custom property (like data type) to the "desired configuration"
jason config, will this be passed through to the next module with that
data. The problem that I see is that the if I get 4838393 from voltage how
do I know in the next module if that is a uint16 or a float32, it would be
best if I could tag voltage with float32?

Not 100% sure how all these properties are sent through to the different
modules yet.

Thanks
Brendan

606simps, 2017-12-15:
Hi Brendan,

Correct me if I misunderstood anything. 

The JSON configuration is for the Modbus module to know what read operations need to be done. Adding any custom key-value pair to the configuration will not influence the output message, since they are handled with different piece of code. Furthermore, the module won't understand what such "datatype" means because there's no code logic handling it and thus will simply ignore that. One easy way to accomplish your goal is to rely on the "Address" field of the output message. Modbus module will always output values per register and by Modbus spec it's uint16.

Take the following for an example.

Let's say you are trying to read two registers as a full voltage value.
```json
{
  "PublishInterval": "2000",
  "SlaveConfigs": {
    "Slave01": {
      "SlaveConnection": "192.168.0.1",
      "HwId": "PowerMeter-0a:01:01:01:01:01",
      "Operations": {
        "Op01": {
          "PollingInterval": "1000",
          "UnitId": "1",
          "StartAddress": "400001",
          "Count": "2",
          "DisplayName": "Voltage"
        }
      }
    }
}
    
```

```json
[
  {
    "DisplayName":"Voltage",
    "HwId":"PowerMeter-0a:01:01:01:01:01",
    "Address":"400001",
    "Value":"19775",
    "SourceTimestamp":"2017-11-17 08:43:50"
  },
  {
    "DisplayName":"Voltage",
    "HwId":"PowerMeter-0a:01:01:01:01:01",
    "Address":"400002",
    "Value":"15650",
    "SourceTimestamp":"2017-11-17 08:43:50"
  }
]
```
In your custom module, you can always combine register 400001 and register 400002 as a float32 or whatever datatype you like.

Thanks,
Stephen

606simps, 2018-03-05:
Close for inactivity. Reopen if you still want to discuss.

thegunshow, 2018-03-05:
Hi

I know it doesnt have any impact when reading or writing from a Modbus
perspective, it is used as schema meta data, this means that when we
process the data later, if the schema "Float32", then this can
automatically be processed. The advantage of doing it this way, is that
variable types are defined in the cloud and instead of raw data being sent
between docker modules (at the edge), processed values are sent.

So if I read 400001 - 400004, is the data, 2 x Float32, 2x Int32 or 1x
Int64. If I define the schema in the cloud, 400001-400002 = Int32 &
400003-400004 = Float32, so I parse Int32 and Float32 between docker
modules.

Schema is a really important part of this process!!

Hope this makes sense

Thanks
Brendan

606simps, 2018-03-07:
Hi Brendan,

I understand your concern. But I am not quite sure how I can modify the module to fit your request. Could you kindly provide a few example so I can look into it.

Thanks,
Stephen

kiranpradeep, 2018-07-18:
@606simps Moving the job of data types to a custom module is leading to a unclean solution. Agree that it keeps iot-edge-modbus module simple. But the overall solution ugly.

Now the edge v2 cloud developer has to write JSON configs for iot-edge-modbus module, and then JSON config for custom module all in **sync**.  Then again, handle all borderlines cases like failure to read one of registers of multi register type.

whendric, 2019-10-26:
Has there been any progress on adding the ability to define the data and byte swap types for an operation? I have upcoming applications that will require being able to do so, looking through some of the v2 preview documentation and code there are references to data types (Float, Int16 and Int32) and swap modes (BigEndian, BigEndianByteSwap, LittleEndian, LittleEndianByteSwap) but no mention of if they're implanted in the 1.0.5 release?

suneetnangia, 2019-10-28:
@whendric guidance is to use v2 for new workloads. We have designed it in a way to allow extensibility of data types etc. as you have figured out.

whendric, 2019-10-28:
Thanks for the quick response!
Couple of questions;
1: Is the published Modbus module in the Azure market place V2? It’s current image URI is mcr.microsoft.com/azureiotedge/Modbus:1.0. If not where would I find the appropriate URI address.
2: What are the definable fields for the JSon config? Is it “datatype” and “swapmode” and what is the correct syntax for the variable?
Thanks
Wayne
    	
    	

    	Get Outlook for iOS

suneetnangia, 2019-10-28:
@whendric 
1. Marketplace version is 1.x, we have not got around to publish v2 to marketplace yet, plan is to do that in near future but that shouldn't block you. You can still compile v2 and create iotedge module/container image from the solution. There's even an Azure DevOps pipeline definition in v2 to get you started.
2. It's defined by "ValueType" which maps to "ModbusValueTypes" Enum in V2 code. More info on this available in unit tests project in v2 solution.
HTH

yphuangms, 2020-10-20:
Close for inactivity.

closed 2020-10-20 by yphuangms