Azure/iot-edge-modbus #18: Polling interval

thegunshow, 2017-12-12:
This seems to have no effect on the module. Jus reads as fast as possible.

Would be good to timestamp the output.

Thanks

606simps, 2017-12-13:
Hi,
Could you provide your configuration?

surabhshah, 2017-12-13:
Hi @thegunshow, are you using the docker image and setting the latest configuration?
If so, then you need to provide the older key for "PollingInterval" i.e. simply "Interval" in the configuration to see the changes, because I guess the public docker image is not yet updated with latest changes. In the meanwhile, you could probably build the source and create a custom docker image and use it.

Thanks,
Surabh

thegunshow, 2017-12-13:
Hi Surabh

I am using the public image so this probably is the problem, I will build
the image from source and see what happens.

Maybe microsoft can update the public image as well?

Thanks
Brendan

606simps, 2017-12-14:
Yes, @surabhshah is correct. I will update the public image soon.

closed 2017-12-14 by 606simps

606simps, 2017-12-14:
Try to use microsoft/azureiotedge-modbus-tcp:1.0-rtu-preview.
This version has Polling Interval working properly and also Modbus RTU functionality.

thegunshow, 2017-12-14:
Thanks, I will give it a try.