Azure/iot-edge-modbus #20: Can tcp port be made configurable?

kiranpradeep, 2017-12-26:
Currently iot-edge-modbus.json doesn't accept TCP port as a parameter. Can this be added?

A non related query: You have mapped modbus RTU devices to inside containers with "Container Create Option Section"([1](https://github.com/Azure/iot-edge-modbus#configure-modbus-rtu)). I tried reading iot-edge docs, but this area is not documented. Could you please point to the docs or sample source from which you go this? Please feel free to not respond to this documentation request if this is not in your scope.

606simps, 2018-01-02:
1. Yes, I can add TCP port as a parameter in next update.
2. The source of "Container Create Option Section" is really from Docker, it is not iot-edge specific. You can find more details [here](https://docs.docker.com/engine/api/v1.30/#operation/ContainerCreate).

606simps, 2018-01-15:
Hi,
Please try to pull the latest source code and rebuild your container image. TCP port is now configurable, default to 502 if not set. Please note that image from Docker Hub is not updated yet, so you'll have to build the image from source code.

Thanks.

kiranpradeep, 2018-07-18:
@606simps Thanks. Can this be made configurable for v1 version as well?

606simps, 2018-07-19:
Hi, yes I can, but we now have higher priority with v2. I will reopen the issue and put on our list.

closed 2020-10-20 by yphuangms

yphuangms, 2020-10-20:
Hi @kiranpradeep , v1 is an obsolete version and it's no longer in maintenance. However, we'll still keep v1 source code in this repo. You are welcomed to fork this repo to customize the code for your need.