Azure/iot-edge-modbus #23: Unable to run as stand-alone mode, because its dependant on ConnectionString to the EdgeHub and EdgeHub Certificate

ilyas-it83, 2018-01-04:
I pretty much followed the documentation and tried executing the module as a standalone app and received the following error message. Basically it looks for Environment variables for IoTEdge Connection string and Certificate File path. Are there any work around for this, if so where do i get these information from.

Error Message
---------------
Unhandled Exception: System.AggregateException: One or more errors occurred. (Value cannot be null.
Parameter name: connectionString) ---> System.ArgumentNullException: Value cannot be null.
Parameter name: connectionString
   at Microsoft.Azure.Devices.Client.DeviceClient.CreateFromConnectionString(String connectionString, ITransportSettings[] transportSettings, IDeviceClientPipelineBuilder pipelineBuilder)
   at Modbus.Containers.Program.<InitEdgeModule>d__11.MoveNext() in C:\Ilyas\AzureFunction\iot-edge-modbus\src\Program.cs:line 108
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.Tasks.Task.Wait()
   at Modbus.Containers.Program.Main(String[] args) in C:\Ilyas\AzureFunction\iot-edge-modbus\src\Program.cs:line 39

606simps, 2018-01-07:
Please follow this [section](https://github.com/Azure/iot-edge-modbus#debug-from-visual-studio-2017) to configure as standalone mode.
This should fix the two issues you encountered.

Thanks.

ilyas-it83, 2018-01-10:
Thanks, it worked. wrt to sending messages from IotHub directly to the modbus module or via another module can be little bit elaborate, similarly message format and payload is not easy to understand. Could you please update the wiki to follow along.

606simps, 2018-01-11:
Could you be specific about how it is not easy to understand? So I know how to explain better in the documents. Thanks.

ilyas-it83, 2018-01-11:
I'm referring to the "write to modbus" section of the document, here detailed instructions on how to send input messages to Modbus TCP when running the solution using Visual Studio as well as IoTEdge Module will help. Apart from this, detailed instructions on using Twin  Properties to change the behavior of the device, Sending and responding to Direct Messages send from IotHub to device, Calling Methods etc.

606simps, 2018-01-12:
Understood. Will add related content. Thanks.