Azure/iot-edge-modbus #24: Combine multiple operation results into one object holding an array of values

sandervandevelde, 2018-01-16:
When multiple operations are executed, the results are sent as an array. But when consumed by Azure Stream Analytics, these seem to be split into separate actions. 

Then, the relationship between the values is lost.

Please put the operations in an array, within a result class.

606simps, 2018-01-17:
Is this the message format you are looking for?
```json
[
  {
    "DisplayName":"Voltage",
    "HwId":"PowerMeter-0a:01:01:01:01:01",
    "SourceTimestamp":"2017-11-17 08:43:50",
    "Values":
      [
        {    
          "Address":"400001",
          "Value":"100"
        },        
        {    
          "Address":"400002",
          "Value":"200"
        }
      ]
  }
]
```

sandervandevelde, 2018-01-17:
I would like to know all related settings (eg. the voltage and current, (and switches etc.) at a specific moment).
Normally, I put everything at the same interval but here is an example with multiple, different, intervals.
(the "PollingTimestamp" could be left out, it can be calculated, relative to the start timestamp and the interval factor between polling and publish)
Could this work?
[{
 "HwId": "PowerMeter-0a:01:01:01:01:01",  // this is the first slave
 "PublishTimestamp" : "2018-01-01T01:01:01.000", // every two seconds
 "Values" : 
 [
 {
   "PollingTimestamp" : "2018-01-01T01:01:01.000" // voltage every second = 2 entries
   "DisplayName" : "Voltage", 
   "StartAddress" : 400001,
   "Value" : 42.0
 }, 
 {
   "PollingTimestamp" : "2018-01-01T01:01:02.000"
   "DisplayName" : "Voltage", 
   "StartAddress" : 400001,
   "Value" : 42.1
 }, 
 {
   "PollingTimestamp" : "2018-01-01T01:01:01.000" // current every 500 ms = 4 entries
   "DisplayName" : "Current", 
   "StartAddress" : 400002,
   "Value" : 43.0
 },
 {
   "PollingTimestamp" : "2018-01-01T01:01:01.500"
   "DisplayName" : "Current", 
   "StartAddress" : 400002,
   "Value" : 43.1
 },
 {
   "PollingTimestamp" : "2018-01-01T01:01:02.000"
   "DisplayName" : "Current", 
   "StartAddress" : 400002,
   "Value" : 43.2
 },
 {
   "PollingTimestamp" : "2018-01-01T01:01:02.500"
   "DisplayName" : "Current", 
   "StartAddress" : 400002,
   "Value" : 43.3
 }
 ]
},
{
 "HwId": "PowerMeter-0a:01:01:01:01:02",  // this is the second slave
 "PublishTimestamp" : "2018-01-01T01:01:01.000", // every two seconds
 "Values" : 
 [ 
    ...
 ] 
}]

606simps, 2018-01-18:
I made a slight modification to your example. Do you think this still cover your needs?
```json
{
  "PublishTimestamp": "2018-01-01T01:01:02.000", // every two seconds
  "Content": [
    {
      "HwId": "PowerMeter-0a:01:01:01:01:01", // this is the first slave
      "Data": [
        {
          "PollingTimestamp": "2018-01-01T01:01:01.000", // voltage every second = 2 entries
          "DisplayName": "Voltage",
          "Values": [
            {
              "Address": "400001",
              "Value": "42.0"
            },
            {
              "Address": "400002",
              "Value": "42.1"
            }
          ]
        },
        {
          "PollingTimestamp": "2018-01-01T01:01:02.000", // second entry of voltage
          "DisplayName": "Voltage",
          "Values": [
            {
              "Address": "400001",
              "Value": "42.2"
            },
            {
              "Address": "400002",
              "Value": "42.3"
            }
          ]
        },
        {
          "PollingTimestamp": "2018-01-01T01:01:01.000", // current every 2 second = 1 entrie
          "DisplayName": "Current",
          "Values": [
            {
              "StartAddress": "400003",
              "Value": "43.1"
            },
            {
              "StartAddress": "400004",
              "Value": "43.2"
            }
          ]
        }
      ]
    },
    {
      "HwId": "PowerMeter-0a:01:01:01:01:02", // this is the second slave
      "Data": [
        ...
      ]
    }
  ]
}
```
1. Within each Modbus slave, grouped values of same "DisplayName" and "PollingTimestamp" in an "Data" array, this could save some space on the message size.
2. Because there could be multiple register "Address" and "Value", so make "Values" an array to hold these.
3. Moved PublishTimestamp outside of the Modbus slave, since this timestamp will be same for all Modbus slaves in the same message.

sandervandevelde, 2018-01-18:
Thanks for the update. 
There is a issue with your example. In my opinion, "voltage" and "current" each represent one 'startaddress' (like 40001 and 40002) (see the configuration example on the readme.md of this repository).

But the update shows both 40001 and 40002 (voltage AND current) under Voltage. And that is not what I expect from the configuration shown.

If my example is read by ASA, it will be splitted in separate messages. I think this makes it possible to separate messages from devices (powermeter1 and powermeter2) with group by etc. [confirmation needed].

606simps, 2018-01-19:
In my example, I have "Voltage" to be address "40001" and "40002" combined, and "Current" to be address "40003" and "40004" combined.

Sometimes a single "Operation" in Modbus module such like "Voltage" and "Current" needs more than one address to represent the value. One Modbus register is 16bits long and is insufficient for data type like float32, int32 or even longer. Some devices (the one power meter I have in hand) use multiple addresses combined to represent one value. So I keep the flexibility for it to have multiple addresses in one "Operation" like "Voltage" and "Current". This is also consistent with the module configuration where you can see a "Count" field under each "Operation". You can still have only one address in your example if that is your scenario, just expect it to come as an array with only one element.

I hope this makes sense. Could you please confirm if my example could be parsed by your ASA properly? It should still be able to split into messages from devices.

Thanks.

sandervandevelde, 2020-04-23:
fixed with pull request.

closed 2020-04-23 by sandervandevelde