Azure/iot-edge-modbus #26: Error when registering EventHandlers

ilyas-it83, 2018-01-19:
I was able to run the module in a stand alone module, but starting yesterday i'm receiving the following error message  "SetInputMessageHandlerAsync for a named output is available for Modules only"  when i hit the below code.

await UpdateStartFromTwin(moduleTwin.Properties.Desired, ioTHubModuleClient);

Whats the reason behind this behavior?

606simps, 2018-01-21:
Because there is a check in the "SetInputMessageHandlerAsync()" API. That is saying the API can only be used when running as an **IoT Edge module**. Check **Program.cs** file at line #188, the "SetInputMessageHandlerAsync()" API should be wrapped with a #if IOT_EDGE/#endif section. You might have accidentally modified the code or not forgot to comment out #define IOT_EDGE at line #1 of **Program.cs**.

606simps, 2018-01-29:
Closing this due to inactivity. Please reopen if issue persists.

closed 2018-01-29 by 606simps