Azure/iot-edge-modbus #28: Runtime error

termocontrole, 2018-02-08:
Hi,
I'm trying to run the module, but I get an error in my IoT Edge Dashboard.

The status of my modbus module is: 

- Runtime status: **backoff** 
- Exit code: **128**

I am trying to use **modbus RTU** and I do not know if I configured **"Container Create Option section"** correctly:
```
{
  "HostConfig": {
    "Devices": [
      {
        "PathOnHost": "user",
        "PathInContainer": "user",
        "CgroupPermissions": "rwm"
      }
    ]
  }
}
```

Where can I look to try to solve this?

606simps, 2018-02-09:
Could you please attach the edgeAgent console output?
```cmd
docker logs edgeAgent
```
What's the platform you are trying to run IoT Edge and Modbus module?
Currently Modbus RTU only supports Linux host + Linux container because of Windows limitation.

termocontrole, 2018-02-09:
Oh man, now make sense lol

I changed that configuration to the path that the device is connect (`/dev/ttyUSB0`) and now the module is running.

I didn't make the connection work yet, but I guess that is wrong config.

Using the command `docker logs modbus -f` I have the output:

```
Added Cert: /mnt/edgemodule/edge-device-ca.cert.pem
IoT Hub module client initialized.
Desired property change:
{"PublishInterval":"6000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyUSB0","HwId":"1","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"5000","UnitId":"1","StartAddress":"30007","Count":"1","DisplayName":"manyVAR"}}}},"$version":6}
Attempt to load configuration: {"PublishInterval":"6000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyUSB0","HwId":"1","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"5000","UnitId":"1","StartAddress":"30007","Count":"1","DisplayName":"manyVAR"}}}},"$version":6}
Opening.../dev/ttyUSB0
```

I will try to change the some configuration on the `properties.desired`, let's see what happen.

I will close the issue when I proper configure this, could be that way?

Thank you very much.

termocontrole, 2018-02-14:
Hi,

I can not get through the "Opening ..." part.

When the connection is established, can I see the output data using `docker logs modbus -f`? Or can the data only be seen in the cloud?

Thank you.

termocontrole, 2018-02-15:
Hi,

I'm trying to run it on a Raspberry Pi. Do you think that could be that?

Thank you.

606simps, 2018-02-20:
Hi,

The module has been verified on Raspberry Pi 3 with Ubuntu Mate. If the connection is established successfully, you can see data using ```docker logs modbus -f ```. I would suggest you to first verify the serial connection with standalone mode or other serial communication software you had.

termocontrole, 2018-02-20:
Hi,

I tested the module with a Modbus TCP/IP device to see if I was not confusing something in the settings and everything worked perfectly.

So I tested my Modbus RTU device with my python script for Modbus and it worked, I also tested the device with Modbus Poll and it ran smoothly.

I tried debugging your code on a desktop with Ubuntu and found that the code is stuck in **[this line](https://github.com/Azure/iot-edge-modbus/blob/a75915a424e6c01667235c5caaedac2f2db3688b/src/Program.cs#L45)**.

The curious thing is that both the TX pin and the RX pin of my RTU device are blinking, indicating some communication, yet no value is displayed.

Any suggestion?

Thank you.

606simps, 2018-02-21:
Hi,

The line you pointed is the main thread of the module which doesn't tell much about the Modbus communication. Could you try to debug into the Modbus thread at line 587 and line 672 of ModbusSlave.cs? Also, what's the configuration you are using? Thanks.

termocontrole, 2018-02-22:
Hi Stephen,

I did as you asked:

On **[line 587 (ModbusSlave.cs)](https://github.com/Azure/iot-edge-modbus/blob/a75915a424e6c01667235c5caaedac2f2db3688b/src/ModbusSlave.cs#L587)** there was no problem, it seems the serial connection was established successfully and the serial port was opened, since there was no error.

In **[line 672 (ModbusSlave.cs)](https://github.com/Azure/iot-edge-modbus/blob/a75915a424e6c01667235c5caaedac2f2db3688b/src/ModbusSlave.cs#L672)**, where the data is read, the code is looped from **[line 672](https://github.com/Azure/iot-edge-modbus/blob/a75915a424e6c01667235c5caaedac2f2db3688b/src/ModbusSlave.cs#L672)** to **[line 706](https://github.com/Azure/iot-edge-modbus/blob/a75915a424e6c01667235c5caaedac2f2db3688b/src/ModbusSlave.cs#L706)**. Studying each line, I realized that the problem occurred within the function **[ReadResponse()](https://github.com/Azure/iot-edge-modbus/blob/a75915a424e6c01667235c5caaedac2f2db3688b/src/ModbusSlave.cs#L707)**, since the `response` always returned **_null_**.

Looking to better understand your code, I debugged the process on **Modbus TCP/IP**, everything went well and I discovered that actually **[ReadResponse()](https://github.com/Azure/iot-edge-modbus/blob/a75915a424e6c01667235c5caaedac2f2db3688b/src/ModbusSlave.cs#L486)** was where the magic happened.

Coming back to **Modbus RTU**, I made some assumptions, one of which is that the **[config.RetryCount property](https://github.com/Azure/iot-edge-modbus/blob/a75915a424e6c01667235c5caaedac2f2db3688b/src/ModbusSlave.cs#L716)** must be greater than zero for reading, correct me if I am wrong. With that, I set the property in `iot-edge-modbus.JSON` and managed to get the code to enter that part, but eventually it got stuck on **[line 211 of the SerialDevice.cs file](https://github.com/Azure/iot-edge-modbus/blob/a75915a424e6c01667235c5caaedac2f2db3688b/src/SerialDevice.cs#L211)**, so I do not know if my guess was correct.

I believe the serial port configuration is correct because the **[m_serialPort.Write(request, 0, reqLen)](https://github.com/Azure/iot-edge-modbus/blob/a75915a424e6c01667235c5caaedac2f2db3688b/src/ModbusSlave.cs#L683)** is executed and the RX/TX LEDs blinks.

Any suggestions to continue the searches?

My **Modbus RTU file**:

```
{
  "PublishInterval": "10000",
  "SlaveConfigs": {
    "Slave01": {
      "SlaveConnection": "/dev/ttyUSB0",
      "HwId": "RPi",
      "RetryCount": "10",
      "BaudRate": "9600",
      "DataBits": "8",
      "StopBits": "1",
      "Parity": "NONE",
      "FlowControl": "NONE",
      "Operations": {
        "Op01": {
          "PollingInterval": "1000",
          "UnitId": "1",
          "StartAddress": "000001",
          "Count": "2",
          "DisplayName": "COIL"
        }
      }
    }
  }
}
```

My **Container Create Option config**:

```
{
  "HostConfig": {
    "Devices": [
      {
        "PathOnHost": "/dev/ttyUSB0",
        "PathInContainer": "/dev/ttyUSB0",
        "CgroupPermissions": "rwm"
      }
    ]
  }
}
```

606simps, 2018-02-23:
Hi,

Yes, you are correct that retryCount should be a positive number. I forgot to put the same validation I did to Modbus TCP at line 53, it is a bug. 

Could you elaborate what you mean by ```it got stuck on line 211 of the SerialDevice.cs file```. Does the program hang at line 211 and never comeback or does it throw exception? Which read does this happen, is it in the ```while (header_len < 3 && retry < config.RetryCount)``` loop or in the ```while (data_len < byte_counts && retry < config.RetryCount)``` loop? If it doesn't fail at the first try, meaning there are some data in the ```response``` array, could you dump them?

termocontrole, 2018-02-23:
Hi @606simps,

No problem, this bug is easy to solve, but maybe it would be fine to be flagged in the documentation. Just a suggestion.

Of course, the program hangs on **[line 211](https://github.com/Azure/iot-edge-modbus/blob/a75915a424e6c01667235c5caaedac2f2db3688b/src/SerialDevice.cs#L211)** `int res = ComWrapper.com_read(fd.Value, readingBuffer, len);` and does nothing else, without throw exception.
It happens already in the first reading loop: `while (header_len <3 && retry <config.RetryCount)`
So, unfortunately, the `response array` is **_null_**.

Thank you.

606simps, 2018-02-26:
Hi,

Sorry for the late response. I tried to duplicate your issue and find this might be because of incorrect serial port settings in ```ComWrapper.c```. I made a quick fix and tested O.K. Could you please pull the latest source code and try your scenario again?

Thanks.

termocontrole, 2018-02-27:
Hi @606simps,

Cool, I'm will test the new code and give you a feedback as soon as I finish.

Thank you very much.

termocontrole, 2018-03-01:
Hi @606simps ,

I tested his code and it almost worked: the reading stopped when I disconnected and connected the device.

I decided to go a little deeper into your code and saw that you are using `termios.h` to make the serial connection on linux.

Testing here, I discovered that the best configuration would be **non-canonical mode**, working in raw mode and removing any treatment by `termios.h`, probably in _canonical mode_ the end-of-line character was expected, but it did not arrive, which blocked the connection inside the buffer.

I preferred to remove `VMIN = 0` and `VTIME = 0` to avoid wasting processing by polling the serial connection.

I am not an expert in `termios.h` but with these assumptions I managed to make the connection work and the data to be sent successfully.

I made a fork of the project and made my main changes available **[here](https://github.com/termocontrole/iot-edge-modbus/commit/aa89d19d9820eaa680ce0bc75f80bd6d8834b404)**

I had difficulty generating the docker image because I forgot I should compile `libcomWrapper.so` for ARM (sorry), so I added the file in the fork: **[libcomWrapper.so](https://github.com/termocontrole/iot-edge-modbus/commit/bb197e349e34d633311ef1dc12c5639577b85c7c)**

Finally I generated a docker image working for Raspberry Pi and made available in Docker Hub: **[image](https://hub.docker.com/r/termocontrole/iot-edge-modbus/)**

Thank you very much for all your help, I hope I can contribute with this experience.

606simps, 2018-03-02:
Hi @termocontrole ,

Thank you for all the effort you made helping debug the Modbus RTU. I honestly had a hard time configuring ```termios.h``` due to lack of experience in it. 

Would you like to contribute your work to the repository? Please send a pull request and I would love to merge your change. And I can also update the image on Docker Hub.

Thanks a lot!

termocontrole, 2018-03-02:
Hi @606simps,

Yeah, `termios.h` is not easy to play with.

Yes, I would like to contribute, I already opened the pull request.

Thank you very much for your patience, I hope I can contribute more with this repo.

closed 2018-03-05 by 606simps