Azure/iot-edge-modbus #29: Desired Properties not getting triggrered

kkc89, 2018-02-26:
Hi all,
With polling interval set to 1 second for 8 sensors with 1 slave. After some time the desired properties does not get triggered(the data does not get received in OnDesiredPropertiesUpdate callback) after changing from the cloud portal. I had to restart the edge using the command sudo iotedgectl stop followed by sudo iotedgectl start in order to receive the desired properties. Any pointers to this issue ?

606simps, 2018-02-26:
Hi,
Could you provide detailed steps to reproduce your issue? I would like to know if the desired properties was never triggered in the module. Or it did get triggered a few times but eventually not receiving anymore.

kkc89, 2018-03-02:
Hi, 
The trigger was getting triggered but misses eventually. 
Anyways Sorry for the trouble as the issue got fixed, Actually another instance of the same edge device was running in a different machine which might have caused this.

closed 2018-03-02 by kkc89

606simps, 2018-03-05:
Hi,
Yes, Azure IoT Device doesn't allow two instance of same device identity connecting to the cloud concurrently. Both device will get connect/disconnect over and over again.