Azure/iot-edge-modbus #30: Docker module updates not getting deployed

kkc89, 2018-03-02:
Hi all,

currently i have deployed the modbus docker image on a linux(ubuntu) edge device. The docker image gets successfully deployed from the container registry created in azure portal. But if any changes are made to the docker image and pushed to the container registry it does'nt gets updated in the edge device. The device keeps running the old docker image. As per my understanding when a updated image is pushed it should get reflected in the edge device immediately.
Is there any configuration i am missing while building the modbus docker image ?

606simps, 2018-03-05:
Does restarting IoT Edge helps? ```iotedgectl restart```
If it doesn't, you might want to rename your docker images with different version tag and also update it from Azure Portal. For example, change from ```microsoft/iotedge-modbus:1.0.0``` to ```microsoft/iotedge-modbus:1.0.1```.

606simps, 2018-03-30:
Closing for inactivity.

closed 2018-03-30 by 606simps

arerlend, 2018-05-29:
@606simps ```iotedgectl restart``` didn't pull the new image for me.  While incrementing the versioning works fine, is there a specific reason why Edge doesn't compare the local image hash to the one in ACR/Docker hub?

606simps, 2018-06-20:
Sorry I didn't see this earlier. This would be a issue more related to the iot edge [repo](https://github.com/Azure/iot-edge), or Docker itself. I had faced the same issue occasionally. Sometimes I would delete my local image, and let edgeAgent pull the latest for me.

arerlend, 2018-06-20:
No worries, just seemed like a missed opportunity for some automated validation.  Thanks for responding.