Azure/iot-edge-modbus #33: How to write to Modbus from IoT Hub

termocontrole, 2018-03-08:
Hi @606simps,

Is there a way to write to Modbus using direct methods or cloud-to-device messages from the IoT Edge panel on the IoT Hub?

I could not make it work. Is it necessary to add some route from the IoT Hub to the Modbus input1?

Thank you.

606simps, 2018-03-09:
Hi @termocontrole ,

IoT Edge currently doesn't support routing message from cloud to device, only routes message from one module to another module. It's the limitation of IoT Edge side, you can get more details asking this question at their community.

Thanks.

termocontrole, 2018-03-09:
Ooohhhh, I do not believe it, I spent almost a whole day reading the documentation of IoT Hub and Edge and had not been able to make it work. Now I understand why lol, at least now I can work on an alternative.

Thank you very much.

closed 2018-03-09 by termocontrole