Azure/iot-edge-modbus #35: Token and TLS handshake

termocontrole, 2018-03-27:
Hi @606simps,

I openned an **[issue](https://github.com/Azure/iot-edge/issues/545)** with the **IoT Edge team** and solved it, but I have **2 doubts** that they said to ask here, please, see if you can help.

All the data that is being sent by the **IoT Edge is received by IoT Hub** (I used **Stream Analytics** to measure that).

Here is my **edgeHub log**:

```
2018-03-27 13:51:03.528 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.MessagingServiceClient] - Sending message for device Id TC-IoT-Edge/modbus
2018-03-27 13:51:03.529 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Routing.RoutingEdgeHub] - Received message from TC-IoT-Edge/modbus
2018-03-27 13:51:03.627 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Storage.MessageStore] - Getting next batch for endpoint iothub starting from 45 with batch size 10.
2018-03-27 13:51:03.628 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Storage.MessageStore] - Obtained next batch for endpoint iothub with batch size 1. Next start offset = 46.
2018-03-27 13:51:03.629 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Storage.MessageStore] - Getting next batch for endpoint iothub starting from 46 with batch size 10.
2018-03-27 13:51:03.629 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Storage.MessageStore] - Obtained next batch for endpoint iothub with batch size 0. Next start offset = 46.
2018-03-27 13:51:03.858 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.CloudProxy] - Sending message for device TC-IoT-Edge/modbus
2018-03-27 13:51:04.998 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.CloudConnection] - Getting new token for TC-IoT-Edge/modbus.
2018-03-27 13:51:04.999 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.CloudConnection] - Existing token not found for TC-IoT-Edge/modbus. Getting new token from the client...
2018-03-27 13:51:05.000 +00:00 [WRN] [ProtocolGateway] - Closing connection for device: TC-IoT-Edge/modbus, Microsoft.Azure.Devices.Edge.Hub.Core.EdgeHubConnectionException: Connection closed for device TC-IoT-Edge/modbus., 
2018-03-27 13:51:05.001 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.MessagingServiceClient] - Disposing MessagingServiceClient for device Id TC-IoT-Edge/modbus because of exception - Microsoft.Azure.Devices.Edge.Hub.Core.EdgeHubConnectionException: Connection closed for device TC-IoT-Edge/modbus.
2018-03-27 13:51:05.002 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.DeviceProxy] - Closing device proxy for device Id TC-IoT-Edge/modbus
2018-03-27 13:51:05.002 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.Core.ConnectionManager] - Device connection removed for device TC-IoT-Edge/modbus
2018-03-27 13:51:05.003 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.EdgeHubConnection] - Updating device TC-IoT-Edge/modbus connection status to Disconnected
2018-03-27 13:51:05.213 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.CloudConnection] - Obtained new token for client TC-IoT-Edge/modbus that expires in 00:59:59.7868522
2018-03-27 13:51:05.213 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.CloudConnection] - Updated cloud connection for client TC-IoT-Edge/modbus
2018-03-27 13:51:05.213 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.Core.ConnectionManager] - New cloud connection created for device TC-IoT-Edge/modbus
2018-03-27 13:51:05.213 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.Core.Authenticator] - Successfully authenticated device TC-IoT-Edge/modbus
2018-03-27 13:51:05.214 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.SasTokenDeviceIdentityProvider] - Successfully generated identity for clientId TC-IoT-Edge/modbus and username HIDDEN
2018-03-27 13:51:05.214 +00:00 [INF] [ProtocolGateway] - ClientAuthenticated, TC-IoT-Edge/modbus, 5c147a81
2018-03-27 13:51:05.214 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.ConnectionManager] - Obtained cloud connection for device TC-IoT-Edge/modbus
2018-03-27 13:51:05.216 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.CloudProxy] - Binding cloud listener for device TC-IoT-Edge/modbus
2018-03-27 13:51:05.216 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.Core.ConnectionManager] - New device connection for device TC-IoT-Edge/modbus
2018-03-27 13:51:05.216 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.EdgeHubConnection] - Updating device TC-IoT-Edge/modbus connection status to Connected
2018-03-27 13:51:05.225 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.Core.Device.DeviceMessageHandler] - Bind device proxy for device TC-IoT-Edge/modbus
2018-03-27 13:51:05.225 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.MessagingServiceClient] - Binding message channel for device Id TC-IoT-Edge/modbus
2018-03-27 13:51:05.226 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.SessionStatePersistenceProvider] - Adding subscription $iothub/twin/res/# for client TC-IoT-Edge/modbus.
2018-03-27 13:51:05.226 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.SessionStatePersistenceProvider] - Adding subscription devices/TC-IoT-Edge/modules/modbus/# for client TC-IoT-Edge/modbus.
2018-03-27 13:51:05.227 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.SessionStatePersistenceProvider] - Adding subscription $iothub/twin/PATCH/properties/desired/# for client TC-IoT-Edge/modbus.
2018-03-27 13:51:05.228 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.SessionStateStoragePersistenceProvider] - Set subscriptions from session state for TC-IoT-Edge/modbus
2018-03-27 13:51:05.244 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.CloudProxy] - Updating reported properties for device TC-IoT-Edge/$edgeHub
2018-03-27 13:51:05.244 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - Successfully sent reported properties to cloud for TC-IoT-Edge/$edgeHub and reported properties version 0
2018-03-27 13:51:05.246 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - Updated cached reported property for TC-IoT-Edge/$edgeHub at reported property version 841 cloudVerified True
2018-03-27 13:51:07.360 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.CloudProxy] - Updating reported properties for device TC-IoT-Edge/$edgeHub
2018-03-27 13:51:07.361 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - Successfully sent reported properties to cloud for TC-IoT-Edge/$edgeHub and reported properties version 0
2018-03-27 13:51:07.365 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.TwinManager] - Updated cached reported property for TC-IoT-Edge/$edgeHub at reported property version 841 cloudVerified True
2018-03-27 13:53:34.732 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Mqtt.MessagingServiceClient] - Sending message for device Id TC-IoT-Edge/modbus
2018-03-27 13:53:34.732 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Routing.RoutingEdgeHub] - Received message from TC-IoT-Edge/modbus
2018-03-27 13:53:34.734 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Storage.MessageStore] - Getting next batch for endpoint iothub starting from 46 with batch size 10.
2018-03-27 13:53:34.735 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Storage.MessageStore] - Obtained next batch for endpoint iothub with batch size 1. Next start offset = 47.
2018-03-27 13:53:34.735 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Storage.MessageStore] - Getting next batch for endpoint iothub starting from 47 with batch size 10.
2018-03-27 13:53:34.736 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Storage.MessageStore] - Obtained next batch for endpoint iothub with batch size 0. Next start offset = 47.
2018-03-27 13:53:34.941 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.CloudProxy] - Sending message for device TC-IoT-Edge/modbus
2018-03-27 13:54:46.035 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Storage.RocksDb.DbStoreProvider] - Starting compaction of stores
2018-03-27 13:54:46.038 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Storage.RocksDb.DbStoreProvider] - Starting compaction of store checkpoints
2018-03-27 13:54:46.070 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Storage.RocksDb.DbStoreProvider] - Starting compaction of store sessions
2018-03-27 13:54:46.089 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Storage.RocksDb.DbStoreProvider] - Starting compaction of store twins
2018-03-27 13:54:46.105 +00:00 [INF] [Microsoft.Azure.Devices.Edge.Storage.RocksDb.DbStoreProvider] - Starting compaction of store iothub
[... cut ...]
2018-03-27 14:51:28.642 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.CloudProxy] - Sending message for device TC-IoT-Edge/modbus
```

If you look at my edgeHub logs, you are going to see that the modbus token is renewed every one hour, because the token is set to expire every one hour, is this default? Can I change the validity of this token somewhere?

Another question is that sometimes appers `TLS handshake failed` but does't seem to cause trouble to the data. Do you have any thoughts about this?

Thank you.

606simps, 2018-03-30:
Hi @termocontrole ,

I updated the original issue you opened at iot-edge repo. Please take a look.

Thanks,
Stephen

termocontrole, 2018-03-30:
Hi @606simps,

Thank you for your support, I reopened the original issue and I will let this one open too, when we solve the doubts I close it.

termocontrole, 2018-04-10:
Hi,

As mentioned in the referenced issue, the modbus token can not be changed, and the `TLS handshake failed` failure is being tracked by the Azure IoT Edge team in this **[issue](https://github.com/Azure/iot-edge/issues/475)**.

Thank you.

closed 2018-04-10 by termocontrole