Azure/iot-edge-modbus #36: Socket issues for larger count values

dtjensen, 2018-03-28:
In the modbus config, if you specify a "count" value larger than 251, socket issues start to show in the logs:

Here is a sample of what I see:
Something wrong with the socket, disposing...
Specified argument was out of the range of valid values.
Parameter name: size
Connection lost, reconnecting...

Values up to 251 are okay.

dtjensen, 2018-03-30:
This is most likely the issue. I am currently trying to read from holding registers, but will need to read from input registers soon.  Forgive my ignorance, but if I want to read an array that is 400 registers, how can I do that across multiple reads and know that the data in the second read has not been updated since the first read? I don't understand how that would work.

606simps, 2018-03-30:
Hi,
You are right, as far as I know there is no way to ensure that. The question is what's the relationship/dependency of the registers so that we have to read them all at once. Please tell more about your scenario, so we can figure out a solution.

Thanks.

606simps, 2018-03-30:
Hi,
What's the size of each point? Is it possible to program X and Y value into one register? Or is it possible to use average value of the points?

yphuangms, 2020-10-20:
Close for inactivity.

closed 2020-10-20 by yphuangms