Azure/iot-edge-modbus #37: MessageResponse could not be found in 'Microsoft.Azure.Devices.Client'

sumongithub, 2018-04-16:
MessageResponse  enum does not exist any of the stable build 'Microsoft.Azure.Devices.Client' NuGate pach=kage

Error message when trying to build-

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0246	The type or namespace name 'MessageResponse' could not be found (are you missing a using directive or an assembly reference?)	iot-edge-modbus	D:\Code\VisualStudio\iot-edge-modbus-master\src\Program.cs	136	Active

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0246	The type or namespace name 'MessageResponse' could not be found (are you missing a using directive or an assembly reference?)	iot-edge-modbus	D:\Code\VisualStudio\iot-edge-modbus-master\src\Program.cs	224	Active
     
Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0103	The name 'MessageResponse' does not exist in the current context	iot-edge-modbus	D:\Code\VisualStudio\iot-edge-modbus-master\src\Program.cs	176	Active

Severity	Code	Description	Project	File	Line	Suppression State
Error	CS0103	The name 'MessageResponse' does not exist in the current context	iot-edge-modbus	D:\Code\VisualStudio\iot-edge-modbus-master\src\Program.cs	227	Active

606simps, 2018-04-17:
Hi,

Which version are you using "Microsoft.Azure.Devices.Client"? This should only be in preview version. Mine is 1.6.0-preview-001.

sumongithub, 2018-04-17:
I'm using latest v1.7.1. version 1.6.0-preview-001 is not available in Nuget repository but available in GitHub

606simps, 2018-04-18:
You would need to check the "Include prerelease" box to see preview versions in the Nuget Package Manager.

606simps, 2018-06-06:
Closing for inactivity.

closed 2018-06-06 by 606simps