Azure/iot-edge-modbus #38: Add synchronous message handling for RTU

dtjensen, 2018-04-20:
When using the RTU variant, the async nature of the read operations make it difficult to manage when there are several read operations defined. The behavior I'm seeing is that the register values for one read operation are in the register locations for another read operation. 

For example, if I have two read ops defined, one that reads 1-50 and one that read 51-100, I have observed that the VALUE for register 1 actually gets returned in register 51.  I have anywhere from 8 - 12 read operations, each reading 50 registers, each with a pooling interval of 1500ms.

I believe that implementing synchronous communication should give more consistent results.

dtjensen, 2018-04-26:
The only thing in the logs was the register/value combinations. So, my logs won't be any help unless you knew the modbus values I was receiving.  It's probably easiest for you to define the same number of read ops I described above, with the same polling timing and see if you can replicate that way.

606simps, 2018-04-30:
Do you have a working setting of control flow for your Modbus device? The serial port communication code is at [here](https://github.com/Azure/iot-edge-modbus/blob/master/src/comWrapper.c). You could try to apply your settings to the code. 
I currently don't have a hardware/simulator with these many number of registers. Will try to get one and investigate deeper.

yphuangms, 2020-10-20:
Close for inactivity.

closed 2020-10-20 by yphuangms