Azure/iot-edge-modbus #39: Single stage build for arm32v7 not working

chriswaters78, 2018-05-04:
I am following the single stage build process for arm32v7:

```
$cd iot-edge-modbus/src/
$dotnet restore
$dotnet build
$dotnet publish -f netcoreapp2.0 -c Release
$cd ../
$docker build --build-arg EXE_DIR=./src/bin/Release/netcoreapp2.0/publish -t "modbus:latest" -f Docker/linux-arm32/Dockerfile .
```

But when I deploy this module I get the following error when trying to use MODBUS RTU operations:

_"Unable to load DLL 'libcomWrapper.so': The specified module or one of its dependencies could not be found"_

The libcomWrapper.so file looks like it has correctly generated and is located in the /usr/lib/ folder on the container.

Any ideas what I am doing wrong?

chriswaters78, 2018-05-04:
Think I have figured this out, gcc is building an incorrect binary for arm32.

If I edit the Dockerfile the first line of the docker file from:

FROM gcc:7 AS build-env

to 

FROM arm32v7/gcc:7 AS build-env

Then I no longer get the DLL load error.

606simps, 2018-05-07:
Thank you for reporting this. Don't know why I didn't get this error while testing it. Will update the document accordingly.

closed 2018-10-03 by 606simps