Azure/iot-edge-modbus #41: Modbus server not receiving messages sent

davebanthorpe, 2018-05-18:
Following your instructions I send the payload to the device (via Device Explorer) but the logs for the modbus module (and the modbus server itself) do not show receipt of that message? How do I determine what is going on?

606simps, 2018-05-21:
Hi Dave,

It is a limit of Azure IoT Edge. You can check out more details [here](https://github.com/Azure/iot-edge/issues/451).

davebanthorpe, 2018-05-22:
S'ok fixed it. It works if I use a local edge function to do the job rather than waiting for the data to get to the cloud Hub.

EricACSIT, 2018-06-22:
Hey Davebanthorpe,
The documentation online now explains how it should be possible to send to the modbus module, however it doesn't appear to be working. What do you mean by local edge function? Is this a command you execute on the device, or a direct method to execute the command through the IoT Edge device?

yphuangms, 2020-10-20:
Close for inactivity.

closed 2020-10-20 by yphuangms