Azure/iot-edge-modbus #42: ArgumentNullException: Value cannot be null after Module twin update

sandervandevelde, 2018-05-22:
Hello, I try to update a Module twin. I got this exception which does not give a good description.

But more serious is that the update is not ignored, the module stop functioning.

    40115: 0
    40116: 0
    40117: 0
    40118: 0
    40119: 0

    Desired property change:
    {"PublishInterval":"2000","SlaveConfigs":{"Slave01":{"SlaveConnection":"10.1.35.104","HwId":"eWONAtcData","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"110","Count":"2","DisplayName":"Dancer22Position"}}}},"$version":4}
    Attempt to load configuration: {"PublishInterval":"2000","SlaveConfigs":{"Slave01":{"SlaveConnection":"10.1.35.104","HwId":"eWONAtcData","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"110","Count":"2","DisplayName":"Dancer22Position"}}}},"$version":4}

    Error when receiving desired property: System.ArgumentNullException: Value cannot be null.
    Parameter name: format

    at System.String.FormatHelper(IFormatProvider provider, String format, ParamsArray args)
    at Modbus.Slaves.ModbusSlaveSession.ProcessResponse(ModbusSlaveConfig config, ReadOperation x) in /app/ModbusSlave.cs:line 235
    at Modbus.Slaves.ModbusSlaveSession.<SingleOperation>d__26.MoveNext() in /app/ModbusSlave.cs:line 194
    --- End of stack trace from previous location where exception was thrown ---
    at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
    at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
    at Modbus.Slaves.ModbusSlaveSession.<>c__DisplayClass21_0.<<ProcessOperations>b__0>d.MoveNext() in /app/ModbusSlave.cs:line 172

606simps, 2018-05-23:
Hi, what's the current version of module and platform you are using? Also, please share your old Twin and new Twin so I can try to repro the issue on my side.

Thanks.

sandervandevelde, 2018-05-29:
This issue is not repeatable again after testing it. Most likely, it was related to the problems with the IoT Edge portal update last week.
It was working again after a redeployment of the Edge configuration (a deletion of the module followed by adding the module).
Thanks for your support.