Azure/iot-edge-modbus #44: Machine restart - no data from modbus to edge

davebanthorpe, 2018-06-06:
HI - I have noticed that sometimes after a reboot of the machine, although the modbus module comes up fine and is reporting data, no data is received at the IoT Hub. It would appear that either data is not arriving at the edgeHub from the module. If I try and send a device twin update the module I get this error in the module logs:

`Error when receiving desired property: Operation timeout expired.`

Restarting the module does not fix it - the only fix seems to be to redeploy the module.

Can you advise how to debug this or advise as to what the issue is?

606simps, 2018-06-06:
Hi,
Can you attach logs from edgeAgent, edgeHub and modbus when the issue happens? Both before and after you redeploy the modules. Also the docker containers' status you get from ***docker ps -a***, and ***docker images*** would be helpful. I'll try to replicate the issue while looking into your logs.

Thanks.

davebanthorpe, 2018-06-08:
OK - sorry for delay, was debugging another edgeHub issue that I thought was related.

Logs attached as requested

Yesterday I started the module at 3pm. This morning after running "docker ps" command it reports an update of all modules of 18 hours which is correct. However, I had a "docker logs -f --tail 20 modbus" window open and that had stopped showing data at around 2am - no errors reported, just stopped reporting data.  Restarting the modbus module got data flowing again.

[dockerPS.txt](https://github.com/Azure/iot-edge-modbus/files/2083564/dockerPS.txt)
[images.txt](https://github.com/Azure/iot-edge-modbus/files/2083565/images.txt)

606simps, 2018-06-14:
Hi,
My observation seems to be slightly different from yours. Now I have the Modbus module running over 24 hours with IoT Edge and still be able to send D2C messages, which are received at IoT Hub properly. However all the modules showed disconnected on the Azure Portal, neither does module twin update work. I found this maybe related to [issue](https://github.com/Azure/iot-edge/issues/471).

yphuangms, 2020-10-20:
Close for inactivity.

closed 2020-10-20 by yphuangms