Azure/iot-edge-modbus #45: Modbus module fails with IoT Edge GA

bryan-starbuck, 2018-06-28:
With the IoT Edge GA release the module fails to run.  

From iotedge logs modbus:
Missing path to certificate collection file.

Unhandled Exception: System.InvalidOperationException: Missing path to certificate file.
   at Modbus.Containers.Program.InstallCert() in /app/Program.cs:line 79
   at Modbus.Containers.Program.Main(String[] args) in /app/Program.cs:line 35
Missing path to certificate collection file.

basimolimajeed, 2018-06-29:
I am having a failure issue with the image deplyed from microsoft/azureiotedge-modbus-tcp:1.0-preview . I get "Runtime Status = backoff" and "Exit Code 139"

chriswaters78, 2018-06-29:
Hi, pretty sure the V2 modules are not supposed to be compatible with GA, none of mine would run without change. I created a new version of the Modbus module by copying the code over to a skeleton GA module created from the dotnet template as per the Quickstart guide.

The main differences are in the Init method code, updated packaged references, and any references to DeviceClient change to ModuleClient.

sandervandevelde, 2018-06-29:
@chriswaters78 Are you able to share your module? Is it put in a public repository? Or are you able to provide it as a pull request to @606simps ?

MaxKhlupnov, 2018-07-02:
I Have the same issue "Runtime Status = backoff" and "Exit Code 139".

606simps, 2018-07-03:
This module is working in progress. Soon a new version supporting IoT Edge GA will be released. Thanks for all your patience and feedbacks.

606simps, 2018-07-03:
Guys, please switch to GA-develop branch and test the module with IoT Edge GA. Open issues if you encounter any. Thanks!

closed 2018-07-03 by 606simps

606simps, 2018-07-04:
Prebuilt images can be found here.
Linux amd64: microsoft/azureiotedge-modbus-tcp:GA-preview-amd64
Linux arm32v7: microsoft/azureiotedge-modbus-tcp:GA-preview-arm32v7