Azure/iot-edge-modbus #48: failure in standalone mode for dotnet application

kiranpradeep, 2018-07-18:
Is it possible to run this dotnet module without iot-edge v2 runtime for debug purpose? I tried running as a normal dotnet application and it failed with below error message. I haven't installed any iot-edge v2 related applications (edgeHub, edgeAgent etc). For dotnet application, is standalone mode not supported?

> Unhandled Exception: System.AggregateException: One or more errors occurred. (Environment variable IOTEDGE_WORKLOADURI is required.) ---> System.InvalidOperationException: Environment variable IOTEDGE_WORKLOADURI is required.
>    at Microsoft.Azure.Devices.Client.Edge.EdgeModuleClientFactory.CreateInternalClientFromEnvironmentAsync()
>    at Modbus.Containers.Program.InitEdgeModule() in /Users/xxxxxx/Desktop/sdaf/iot-edge-modbus-master/iotedgeModbus/modules/iotedgeModbus/Program.cs:line 69
>    --- End of inner exception stack trace ---
>    at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
>    at System.Threading.Tasks.Task.Wait()
>    at Modbus.Containers.Program.Main(String[] args) in /Users/xxxxxxx/Desktop/sdaf/iot-edge-modbus-master/iotedgeModbus/modules/iotedgeModbus/Program.cs:line 32
> Abort trap: 6

606simps, 2018-07-19:
We recently removed standalone mode because now iot-edge v2 has support [debugging into modules](https://docs.microsoft.com/en-us/azure/iot-edge/how-to-develop-csharp-module). The modbus module is designed to work with iot-edge. If you only want to verify connection with modbus device, there are a plenty of other free software can do this. If you really want the standalone mode, you can still modify the code to achieve this.

kiranpradeep, 2018-07-20:
@606simps Thanks for the reply. README.md gave an impression like standalone is supported. Can you please update README.md.

closed 2018-07-20 by kiranpradeep

606simps, 2018-07-20:
Thank you. REAME.md is updated.

kiranpradeep, 2018-07-23:
@606simps On a second thought, a standalone mode really helps while reporting issues. I am now reporting issues on v1 only since it is easily reproducible in v1. I feel the same issues may exist in v2, but it is hard to test v2 with need for Docker and v2 agents etc.