Azure/iot-edge-modbus #55: Changing RTU StartAddress can result in ArgumentNullException

sandervandevelde, 2018-08-27:
I have the following module configured for Modbus RTU and it is running as expected (receiving values):

uno2271gsv@uno2271gsv-UNO-2271G-E2xAE:~$ docker ps
CONTAINER ID        IMAGE                                                             COMMAND                   CREATED             STATUS              PORTS                                                                  NAMES
aee3e18ee461        microsoft/azureiotedge-modbus-tcp:0.0.1-amd64                     "dotnet iotedgeModbu…"    9 minutes ago       Up 3 minutes  

I have the following ModuleTwin:

>     "desired": {
>      "PublishInterval": "10000",
>      "SlaveConfigs": {
>        "Slave01": {
>          "SlaveConnection": "/dev/ttyS0",
>          "BaudRate": "9600",
>          "DataBits": "8",
>          "StopBits": "1",
>          "Parity": "NONE",
>          "FlowControl": "NONE",
>          "HwId": "Adam4168U1",
>          "Operations": {
>            "Op01": {
>              "PollingInterval": "10000",
>              "UnitId": "1",
>              "StartAddress": "00017",
>              "Count": "8",
>              "DisplayName": "Relays"
>            }
>          }
>        }
>      },
>      "$metadata": {

This is a part of the output:

> 00020: 0
> 00021: 0
> 00022: 0
> 00023: 0
> 00024: 0

I change "00017" into 17.

The module picks up the change and start reading...

> Desired property change:
> {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","HwId":"Adam4168U1AA","Operations":{"Op01":{"PollingInterval":"5000","UnitId":"1","StartAddress":"17","Count":"8","DisplayName":"RelaysAA"}}}},"$version":21}
> Attempt to load configuration: {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","HwId":"Adam4168U1AA","Operations":{"Op01":{"PollingInterval":"5000","UnitId":"1","StartAddress":"17","Count":"8","DisplayName":"RelaysAA"}}}},"$version":21}
> Invalid TcpPort: , set to DefaultTcpPort: 502
> Invalid RetryCount: , set to DefaultRetryCount: 10
> Invalid RetryInterval: , set to DefaultRetryInterval: 50
> Empty CorrelationId: , set to DefaultCorrelationId: DefaultCorrelationId
> Opening.../dev/ttyS0

No data is received as expected. Let's change it back to "00017":

> Error when receiving desired property: System.ArgumentNullException: Value cannot be null.
> Parameter name: format
>    at System.String.FormatHelper(IFormatProvider provider, String format, ParamsArray args)
>    at Modbus.Slaves.ModbusSlaveSession.ProcessResponse(ModbusSlaveConfig config, ReadOperation x) in /app/ModbusSlave.cs:line 238
>    at Modbus.Slaves.ModbusSlaveSession.<SingleOperation>d__27.MoveNext() in /app/ModbusSlave.cs:line 197
> --- End of stack trace from previous location where exception was thrown ---
>    at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
>    at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
>    at Modbus.Slaves.ModbusSlaveSession.<>c__DisplayClass20_0.<<ProcessOperations>b__0>d.MoveNext() in /app/ModbusSlave.cs:line 163

The module freezes. onlt a "docker kill rtu" restarts the module. And because the devicetwin is correct, it starts reading directly.

I also tried "00017" to "00018" but this works ok.

Bonus: in TCP, I also changed "00017" into "17" (I did not touch 00001, 00002, 00018, 40001, 40002). Service keeps running with this continuous message:

> 00018: 0
> 40001: 0
> 40002: 0
> 00001: 1
> 00002: 1
> Modbus exception code: 2

After setting the TCP coil address back to "00017":

> 00002: 1
> Modbus exception code: 2
> Desired property change:
> {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"192.168.1.142","HwId":"Wise4012E-142","Operations":{"Op01":{"PollingInterval":"5000","UnitId":"1","StartAddress":"40001","Count":"1","DisplayName":"KnobOne"},"Op02":{"PollingInterval":"5000","UnitId":"1","StartAddress":"40002","Count":"1","DisplayName":"KnobTwo"},"Op03":{"PollingInterval":"5000","UnitId":"1","StartAddress":"00001","Count":"1","DisplayName":"SwitchOne"},"Op04":{"PollingInterval":"5000","UnitId":"1","StartAddress":"00002","Count":"1","DisplayName":"SwitchTwo"},"Op05":{"PollingInterval":"5000","UnitId":"1","StartAddress":"00017","Count":"1","DisplayName":"RelayOne"},"Op06":{"PollingInterval":"5000","UnitId":"1","StartAddress":"00018","Count":"1","DisplayName":"RelayTwo"}}}},"$version":6}
> Attempt to load configuration: {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"192.168.1.142","HwId":"Wise4012E-142","Operations":{"Op01":{"PollingInterval":"5000","UnitId":"1","StartAddress":"40001","Count":"1","DisplayName":"KnobOne"},"Op02":{"PollingInterval":"5000","UnitId":"1","StartAddress":"40002","Count":"1","DisplayName":"KnobTwo"},"Op03":{"PollingInterval":"5000","UnitId":"1","StartAddress":"00001","Count":"1","DisplayName":"SwitchOne"},"Op04":{"PollingInterval":"5000","UnitId":"1","StartAddress":"00002","Count":"1","DisplayName":"SwitchTwo"},"Op05":{"PollingInterval":"5000","UnitId":"1","StartAddress":"00017","Count":"1","DisplayName":"RelayOne"},"Op06":{"PollingInterval":"5000","UnitId":"1","StartAddress":"00018","Count":"1","DisplayName":"RelayTwo"}}}},"$version":6}
> Invalid TcpPort: , set to DefaultTcpPort: 502
> Invalid RetryCount: , set to DefaultRetryCount: 10
> Invalid RetryInterval: , set to DefaultRetryInterval: 50
> Empty CorrelationId: , set to DefaultCorrelationId: DefaultCorrelationId
> Empty CorrelationId: , set to DefaultCorrelationId: DefaultCorrelationId
> Empty CorrelationId: , set to DefaultCorrelationId: DefaultCorrelationId
> Empty CorrelationId: , set to DefaultCorrelationId: DefaultCorrelationId
> Empty CorrelationId: , set to DefaultCorrelationId: DefaultCorrelationId
> Empty CorrelationId: , set to DefaultCorrelationId: DefaultCorrelationId
> 40001: 0
> 40002: 0
> 00001: 1
> 00018: 0
> 00002: 1
> 00017: 0

docker version:

> Client:
>  Version:      18.06.0-dev
>  API version:  1.37
>  Go version:   go1.10.2
>  Git commit:   daf021fe
>  Built:        Mon Jun 25 21:07:53 2018
>  OS/Arch:      linux/amd64
>  Experimental: false
>  Orchestrator: swarm
> 
> Server:
>  Engine:
>   Version:      dev
>   API version:  1.37 (minimum version 1.12)
>   Go version:   go1.10.2
>   Git commit:   9aac187
>   Built:        Mon Jun 25 20:43:05 2018
>   OS/Arch:      linux/amd64
>   Experimental: false
> uno2271gsv@uno2271gsv-UNO-2271G-E2xAE:~$

Conclusion: Modbus TCP does not return an exception and ignores an invalid address. Modbus RTU gives an exception and seems to be locked.

606simps, 2018-08-28:
Hi Sander,

I tried to replicate your issue on my side, however I didn't see the null exception you saw.

```log
Desired property change:
{"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"00017","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}}}},"$version":84}
Attempt to load configuration: {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"00017","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}}}},"$version":84}
Invalid TcpPort: , set to DefaultTcpPort: 502
Invalid RetryCount: , set to DefaultRetryCount: 10
Invalid RetryInterval: , set to DefaultRetryInterval: 50
Opening.../dev/ttyS0
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Desired property change:
{"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"17","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}}}},"$version":85}
Attempt to load configuration: {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"17","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}}}},"$version":85}
Invalid TcpPort: , set to DefaultTcpPort: 502
Invalid RetryCount: , set to DefaultRetryCount: 10
Invalid RetryInterval: , set to DefaultRetryInterval: 50
Opening.../dev/ttyS0
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Desired property change:
{"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"00017","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}}}},"$version":86}
Attempt to load configuration: {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"00017","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}}}},"$version":86}
Invalid TcpPort: , set to DefaultTcpPort: 502
Invalid RetryCount: , set to DefaultRetryCount: 10
Invalid RetryInterval: , set to DefaultRetryInterval: 50
Opening.../dev/ttyS0
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
Modbus exception code: 1
```
I will add extra check while loading the configuration, if the starting address is not specified with proper 5 digit or 6 digit format, the operation will be ignored.

606simps, 2018-08-28:
update testing images as follow.
microsoft/azureiotedge-modbus-tcp:0.0.1-amd64
microsoft/azureiotedge-modbus-tcp:0.0.1-arm32v7

Please help to validate if the issue is resolved.
Thanks!
```log
Desired property change:
{"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","TcpPort":"502","RetryCount":"10","RetryInterval":"100","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"17","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}}}},"$version":89}
Attempt to load configuration: {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","TcpPort":"502","RetryCount":"10","RetryInterval":"100","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"17","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}}}},"$version":89}
Op01 has invalid StartAddress 17, remove from dictionary...
Opening.../dev/ttyS0
No valid operation in modbus session available!!
Desired property change:
{"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","TcpPort":"502","RetryCount":"10","RetryInterval":"100","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"17","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"},"Op02":{"PollingInterval":"2000","UnitId":"1","StartAddress":"400020","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}}}},"$version":90}
Attempt to load configuration: {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","TcpPort":"502","RetryCount":"10","RetryInterval":"100","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"17","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"},"Op02":{"PollingInterval":"2000","UnitId":"1","StartAddress":"400020","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}}}},"$version":90}
Op01 has invalid StartAddress 17, remove from dictionary...
Opening.../dev/ttyS0
400020: 1000

400021: 0

400020: 1000

400021: 0

400020: 1000

400021: 0

400020: 1000

400021: 0

400020: 1000

400021: 0

400020: 1000

400021: 0

400020: 1000

400021: 0

400020: 1000

400021: 0

400020: 1000

400021: 0

400020: 1000

400021: 0

Desired property change:
{"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","TcpPort":"502","RetryCount":"10","RetryInterval":"100","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"4001","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"},"Op02":{"PollingInterval":"2000","UnitId":"1","StartAddress":"400020","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}}}},"$version":91}
Attempt to load configuration: {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","TcpPort":"502","RetryCount":"10","RetryInterval":"100","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"2000","UnitId":"1","StartAddress":"4001","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"},"Op02":{"PollingInterval":"2000","UnitId":"1","StartAddress":"400020","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}}}},"$version":91}
Op01 has invalid StartAddress 4001, remove from dictionary...
Opening.../dev/ttyS0
400020: 1000

400021: 0

400020: 1000

400021: 0
[... 44 lines cut ...]
400021: 0

400020: 1000

400021: 0

Desired property change:
{"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op02":{"PollingInterval":"2000","UnitId":"1","StartAddress":"400020","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}},"TcpPort":null,"RetryCount":null,"RetryInterval":null}},"$version":93}
Attempt to load configuration: {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op02":{"PollingInterval":"2000","UnitId":"1","StartAddress":"400020","Count":"2","CorrelationId":"MessageType1","DisplayName":"Voltage"}},"TcpPort":null,"RetryCount":null,"RetryInterval":null}},"$version":93}
Invalid TcpPort: , set to DefaultTcpPort: 502
Invalid RetryCount: , set to DefaultRetryCount: 10
Invalid RetryInterval: , set to DefaultRetryInterval: 50
Opening.../dev/ttyS0
400020: 1000

400021: 0

400020: 1000

400021: 0

400020: 1000

400021: 0
```

sandervandevelde, 2018-08-28:
The updated module microsoft/azureiotedge-modbus-tcp:0.0.1-amd64 fixes my issue with the invalid start address.

I used thin device twin (as seen above, I added TcpPort, RetryCount, RetryInterval and CorrelationID to prevent the info lines).

>    "desired": {
>      "PublishInterval": "5000",
>      "SlaveConfigs": {
>        "Slave01": {
>          "SlaveConnection": "/dev/ttyS0",
>          "BaudRate": "9600",
>          "DataBits": "8",
>          "StopBits": "1",
>          "Parity": "NONE",
>          "FlowControl": "NONE",
>          "HwId": "Adam4168U1",
>          "Operations": {
>            "Op01": {
>              "PollingInterval": "5000",
>              "UnitId": "1",
>              "StartAddress": "00017",
>              "Count": "8",
>              "DisplayName": "Relays",
>              "CorrelationId": "Message8Coils"
>            }
>          },
>          "TcpPort": "502",
>          "RetryCount": "10",
>          "RetryInterval": "100"
>        }
>      }

When theis Module Twin is sent, is results in a good response. I get the values of the coils:

> 8/28/2018 10:47:47 PM> Device: [linux6900], Data:[{"machine":{"temperature":101.08185656174635,"pressure":10.123249481717938},"ambient":{"temperature":20.733080430064852,"humidity":25},"timeCreated":"2018-08-28T20:47:47.9496227Z"}]
>8/28/2018 10:47:52 PM> Device: [linux6900], Data:[{"PublishTimestamp":"2018-08-28 20:47:52","Content":[{"HwId":"Adam4168U1","Data":[{"CorrelationId":"Message8Coils","SourceTimestamp":"2018-08-28 20:47:50","Values":[{"DisplayName":"Relays","Address":"00017","Value":"1"},{"DisplayName":"Relays","Address":"00018","Value":"0"},{"DisplayName":"Relays","Address":"00019","Value":"0"},{"DisplayName":"Relays","Address":"00020","Value":"0"},{"DisplayName":"Relays","Address":"00021","Value":"0"},{"DisplayName":"Relays","Address":"00022","Value":"0"},{"DisplayName":"Relays","Address":"00023","Value":"0"},{"DisplayName":"Relays","Address":"00024","Value":"0"}]}]}]}]Properties:
> 'content-type': 'application/edge-modbus-json'

So I change the 00017 into 17:

>00024: 0
>Desired property change:
>{"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","HwId":"Adam4168U1","Operations":{"Op01":{"PollingInterval":"5000","UnitId":"1","StartAddress":"17","Count":"8","DisplayName":"Relays","CorrelationId":"Message8Coils"}},"TcpPort":"502","RetryCount":"10","RetryInterval":"100"}},"$version":41}
>Attempt to load configuration: {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","HwId":"Adam4168U1","Operations":{"Op01":{"PollingInterval":"5000","UnitId":"1","StartAddress":"17","Count":"8","DisplayName":"Relays","CorrelationId":"Message8Coils"}},"TcpPort":"502","RetryCount":"10","RetryInterval":"100"}},"$version":41}
>Op01 has invalid StartAddress 17, remove from dictionary...
>Opening.../dev/ttyS0
>No valid operation in modbus session available!!

The response is ok. 17 is rejected.

So I switch back to 00017:

> Desired property change:
> {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","HwId":"Adam4168U1","Operations":{"Op01":{"PollingInterval":"5000","UnitId":"1","StartAddress":"00017","Count":"8","DisplayName":"Relays","CorrelationId":"Message8Coils"}},"TcpPort":"502","RetryCount":"10","RetryInterval":"100"}},"$version":42}
> Attempt to load configuration: {"PublishInterval":"5000","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyS0","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","HwId":"Adam4168U1","Operations":{"Op01":{"PollingInterval":"5000","UnitId":"1","StartAddress":"00017","Count":"8","DisplayName":"Relays","CorrelationId":"Message8Coils"}},"TcpPort":"502","RetryCount":"10","RetryInterval":"100"}},"$version":42}
> Opening.../dev/ttyS0
> 00017: 1
> 00018: 0
> 00019: 0
> 00020: 0
> 00021: 0
> 00022: 0
> 00023: 0

And I get a gook response with valid values.

I do not have any issues anymore. The response is clear.

closed 2018-09-11 by 606simps