Azure/iot-edge-modbus #57: Unable to deploy on Windows 10 1803 with Windows containers

genezavadski, 2018-09-14:
I have Docker running on Win 10 desktop 1803, configured with windows containers. Docker pull fails with the following:

docker pull mcr.microsoft.com/azureiotedge/modbus:1.0
1.0: Pulling from azureiotedge/modbus
no matching manifest for unknown in the manifest list entries

Iot edge is running successfully, and auto-deployed tempSensor without any issues.

NAME             STATUS           DESCRIPTION      CONFIG
tempSensor       running          Up 9 minutes     mcr.microsoft.com/azureiotedge-simulated-temperature-sensor:1.0
edgeHub          running          Up an hour       mcr.microsoft.com/azureiotedge-hub:1.0
edgeAgent        running          Up an hour       mcr.microsoft.com/azureiotedge-agent:1.0

606simps, 2018-09-27:
Hi, it should be available now, please try again.

606simps, 2018-10-02:
Close. Please reopen if you still having trouble pulling windows container.

closed 2018-10-02 by 606simps