Azure/iot-edge-modbus #60: Error: Invalid TcpPort: , set to DefaultTcpPort: 502

SarithaMaran, 2018-10-08:
## Error: Invalid TcpPort: , set to DefaultTcpPort: 502. Even-though I am trying for RTU device.
I am trying to add my modbus RTU device and following the steps mentioned in README.md provided [here](https://github.com/Azure/iot-edge-modbus).

So what is happening here, it shows a error and is stuck there no telemetry is coming and published to cloud. 
Logs:
```
pi@raspberrypi:~ $ sudo iotedge logs -f modbus --tail=20
IoT Hub module client initialized.
Desired property change:
{"PublishInterval":"2000","Version":"2","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyACM0","HwId":"LGP5310","RetryCount":"10","RetryInterval":"50","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000156","Count":"2","DisplayName":"frequency_hz","CorrelationId":"MessageType1"},"Op02":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000126","Count":"2","DisplayName":"va_r_phase_v","CorrelationId":"MessageType1"},"Op03":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000128","Count":"2","DisplayName":"va_y_phase_v","CorrelationId":"MessageType1"},"Op04":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000130","Count":"2","DisplayName":"va_b_phase_v","CorrelationId":"MessageType1"},"Op05":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000150","Count":"2","DisplayName":"current_r_phase_amp","CorrelationId":"MessageType1"},"Op06":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000152","Count":"2","DisplayName":"current_y_phase_amp","CorrelationId":"MessageType1"},"Op07":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000154","Count":"2","DisplayName":"current_b_phase_amp","CorrelationId":"MessageType1"}}}},"$version":4}
Attempt to load configuration: {"PublishInterval":"2000","Version":"2","SlaveConfigs":{"Slave01":{"SlaveConnection":"/dev/ttyACM0","HwId":"LGP5310","RetryCount":"10","RetryInterval":"50","BaudRate":"9600","DataBits":"8","StopBits":"1","Parity":"NONE","FlowControl":"NONE","Operations":{"Op01":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000156","Count":"2","DisplayName":"frequency_hz","CorrelationId":"MessageType1"},"Op02":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000126","Count":"2","DisplayName":"va_r_phase_v","CorrelationId":"MessageType1"},"Op03":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000128","Count":"2","DisplayName":"va_y_phase_v","CorrelationId":"MessageType1"},"Op04":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000130","Count":"2","DisplayName":"va_b_phase_v","CorrelationId":"MessageType1"},"Op05":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000150","Count":"2","DisplayName":"current_r_phase_amp","CorrelationId":"MessageType1"},"Op06":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000152","Count":"2","DisplayName":"current_y_phase_amp","CorrelationId":"MessageType1"},"Op07":{"PollingInterval":"1000","UnitId":"1","StartAddress":"000154","Count":"2","DisplayName":"current_b_phase_amp","CorrelationId":"MessageType1"}}}},"$version":4}
Invalid TcpPort: , set to DefaultTcpPort: 502
Opening.../dev/ttyACM0
Saving reported properties: {"PublishInterval":2000,"SlaveConfigs":{"Slave01":{"Operations":{"Op01":{"PollingInterval":1000,"UnitId":1,"StartAddress":"000156","Count":2,"DisplayName":"frequency_hz","CorrelationId":"MessageType1"},"Op02":{"PollingInterval":1000,"UnitId":1,"StartAddress":"000126","Count":2,"DisplayName":"va_r_phase_v","CorrelationId":"MessageType1"},"Op03":{"PollingInterval":1000,"UnitId":1,"StartAddress":"000128","Count":2,"DisplayName":"va_y_phase_v","CorrelationId":"MessageType1"},"Op04":{"PollingInterval":1000,"UnitId":1,"StartAddress":"000130","Count":2,"DisplayName":"va_b_phase_v","CorrelationId":"MessageType1"},"Op05":{"PollingInterval":1000,"UnitId":1,"StartAddress":"000150","Count":2,"DisplayName":"current_r_phase_amp","CorrelationId":"MessageType1"},"Op06":{"PollingInterval":1000,"UnitId":1,"StartAddress":"000152","Count":2,"DisplayName":"current_y_phase_amp","CorrelationId":"MessageType1"},"Op07":{"PollingInterval":1000,"UnitId":1,"StartAddress":"000154","Count":2,"DisplayName":"current_b_phase_amp","CorrelationId":"MessageType1"}},"SlaveConnection":"/dev/ttyACM0","RetryCount":10,"RetryInterval":50,"TcpPort":502,"HwId":"LGP5310","BaudRate":9600,"StopBits":1,"DataBits":8,"Parity":0}}}

```
  And no logs after that. Why it is showing "Invalid TcpPort: , set to DefaultTcpPort: 502". How to differentiate between RTU and TCP device? What am I doing wrong here?

606simps, 2018-10-10:
If you see "Opening.../dev/ttyACM0", then you are running in RTU mode and can ignore "set to DefaultTcpPort" message. It's more likely you didn't have serial port configured properly. Please attach your configuration and also try to verify that the serial port is available inside the container.

606simps, 2018-10-11:
Do you have other third party tools that could verified your serial port configuration? I.e baud rate, parity bit..., try to run it on host machine first.

yphuangms, 2020-10-20:
Close for inactivity.

closed 2020-10-20 by yphuangms