Azure/iot-edge-modbus #63: Please name the root array

terrymandin, 2019-01-26:
It is difficult to work with both versions of the output JSON in Azure Stream Analytics as the root array is not named.  For example, see this documentation that assumes "the input events have a property named "arrayField" that is an array datatype":  [Parse JSON and Avro data in Azure Stream Analytics](https://docs.microsoft.com/en-gb/azure/stream-analytics/stream-analytics-parsing-json?toc=%2Fen-us%2Fstream-analytics-query%2Ftoc.json&bc=%2Fen-us%2Fstream-analytics-query%2Fbreadcrumb%2Ftoc.json)

yphuangms, 2019-02-22:
Azure Stream Analytics on IoT Edge now support C# UDF (user defined function) to provide custom defined function to be sued with SQL query. It would worth a try to use C# UDF to process the message of Modbus to any preferred output format.

sandervandevelde, 2019-03-25:
Support for StreamAnalytics is supported using the ' "version":1 ' parameter. This is also known as the "PP (Public Preview)" output.

closed 2020-10-20 by yphuangms