Azure/iot-edge-modbus #71: Module ignores SlaveConnections with a non-"tty" device name

mschrupp, 2019-05-13:
Hi,

so I just found out the modbus module seems to ignore device names specified in a `SlaveConnection` that don't have a name like `/dev/tty*`.

When using device mounts, one can specify the name inside the container. This can be something like `/dev/modbus`.

As it is possible to specify the device here, it shouldn't follow some kind of invisible regex pattern.

Also there is no error at all which hints at this problem.

Thanks!

yphuangms, 2019-05-17:
There are serial device name assumptions in the code. For Linux, the device name should has prefix like "/dev/tty", and for Windows, the device name should be prefixed with "COM". 
These assumptions are to restrict possible input errors in device name.
We understand that applying an alias name to device is quite common and possible. 
We'll consider this suggestion. Thanks!

mschrupp, 2019-05-17:
@yphuangms may I suggest to at least add it to documentation / log an error?

michaelfeil, 2020-05-17:
I assume you have the same inquiry. I would like to connect this serial USB device, which is ignored due to "invalid tcp port"

/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0K5DLA-if00-port0

yphuangms, 2020-10-20:
Close for inactivity.

closed 2020-10-20 by yphuangms