Azure/iot-edge-modbus #75: Communication error

nouira, 2019-06-18:
I'm testing themodule on Iot Edge Simulator and I have this error
`modbus        | Connection lost, reconnecting...
modbus        | Connect Slave failed
modbus        | Connection refused
modbus        | Connection lost, reconnecting...`

Is this due to bridge network of edgeHubDev ?
Is there a configuration to due to overcome that ?

Thanks.

yphuangms, 2019-09-25:
Hi, sorry for the late reply. What's the type of Modbus connection, is it via TCP or via serial interface?

anfelbeghoura, 2019-12-03:
hey; I have the same issue; and I'm using TCP (Ethernet interface). Thank you

nouira, 2019-12-03:
It's late but my problem was due to Azure IoT EdgeHub Dev.
Since then, the version 0.11 has fixed this issue.

closed 2019-12-03 by nouira