Azure/iot-edge-modbus #80: Exception Code 2 - StartAddress being passed as 0

fetapsomi, 2019-09-24:
While trying to get modbus data over TCP I kept getting Exception 2.  

The XML looked fine  as below.  Pretty much exactly as the example provided https://docs.microsoft.com/en-us/azure/iot-edge/deploy-modbus-gateway.

          "Op01": {
            "PollingInterval": "30000",
            "UnitId": "1",
            "StartAddress": "40001",
            "Count": "1",
            "DisplayName": "Voltage"
          }

I am running Ubuntu 18.04 on a hyperv VM.

I installed modbus testing software on a Windows 10 VM and successfully connected to the modbus device.  

A packet trace showed the the successful request from the Win10 VM passed the correct StartAddress but the edge gateway request passed a value of 0 as the StartAddress.  

Any assistance would be appreciated.

yphuangms, 2019-09-25:
Hi, could you capture the logs which containing the exception messages?

fetapsomi, 2019-09-25:
Hi, thanks for the reply.
I've not done anything with Modbus before so apologies in advance. From what I understand, 40001 contains data, Uint32. 
```
40001 | 40002 | 2 | R | 0x03 | SID | Well-known value....
```
I've used a tool, modpol.exe to request data starting at the same register and it returned data, 5 registers.
```
modpoll.exe -c 5 -r 40001 -m tcp 192.168.x.x
modpoll 3.6 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2018 proconX Pty Ltd
Visit https://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP
Slave configuration...: address = 1, start reference = 40001, count = 5
Communication.........: 192.168.x.x, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave... (Ctrl-C to stop)
[40001]: 21365
[40002]: 28243
[40003]: 1
[40004]: 65
[40005]: 18034
```
I changed the register to 40098, float, still have the same problem.
Below is the config of the modbus module, pretty much the same as the example except for the IP and the register.
```
{
  "properties.desired": {
    "PublishInterval": "60000",
    "SlaveConfigs": {
      "Slave01": {
        "SlaveConnection": "192.168.x.x",
        "HwId": "PowerMeter-0a:01:01:01:01:01",
        "Operations": {
          "Op01": {
            "PollingInterval": "30000",
            "UnitId": "1",
            "StartAddress": "40001",
            "Count": "2",
            "DisplayName": "Voltage"
          }
        }
      }
    }
  }
}
```
The saved properties according to the modbus module
```
Saving reported properties: {"PublishInterval":60000,"SlaveConfigs":{"Slave01":{"Operations":{"Op01":{"PollingInterval":30000,"UnitId":1,"StartAddress":"40001","Count":2,"DisplayName":"Voltage","CorrelationId":"DefaultCorrelationId"}},"SlaveConnection":"192.168.x.x","RetryCount":10,"RetryInterval":50,"TcpPort":502,"HwId":"PowerMeter-0a:01:01:01:01:01","BaudRate":null,"StopBits":null,"DataBits":null,"Parity":null}}}
```
And the exception
```
Modbus exception code: 2
```
The successful trace results from modpol.exe shows the starting register as 40000
[image]
And the unsuccessful trace results from modbus module shows the starting address as 0
[image]

Thanks again

yphuangms, 2019-09-25:
Based on your master simulator tool "modpoll" document, I think you misunderstand how to mapping "reference number" to Modbus module configuration "StartingAddress".
When use "modpoll", with "-r 40000" only, which means to use default operation type of "-t 4          16-bit output (holding) register data type (default)". 
The full compatible command line should be :
`modpoll.exe -c 5 -r 40001 -t 4 -a 1 -m tcp 192.168.x.x`
To map to the Modbus module configuration, "-a 1" maps to "UnitId": "1";  "-t 4" means to read from holding register, which is function code 3, so the StartingAddress will start with digit 4; "-r 40001" means to read holding register from address 40001, and combining with the function code, the starting digit will be 4, so it maps to "StartingAddress": "440001".
You should configure Modbus module like this for your test case:

"Op01": {
    "PollingInterval": "30000",
    "UnitId": "1",
    "StartAddress": "440001",
    "Count": "2",
    "DisplayName": "Voltage"
}

fetapsomi, 2019-09-25:
Thanks, that sorted it.  Appreciate your help!

closed 2019-09-26 by yphuangms