Azure/iot-edge-modbus #82: Update to new iotedge version

Dolphinsimon, 2019-12-09:
There is a bug in the 1.0.5 edgeHub. 
 [Could not start module edgeHub](https://github.com/Azure/iotedge/issues/774)
The 1.0.5 is not working with latest iotedge runtime. Almost one year past, there should be a newer version for this module.

sandervandevelde, 2020-05-05:
Although this module is based on older NuGet packages, it runs well on newer IoT Edge Runtimes (even on 1.0.8/1.0.9). So you are not limited to the 1.0.5 runtime for this module.
Can you give more elaborate information about the issues you see? What is the hardware/container configuration you use?

Dolphinsimon, 2020-05-07:
We have tested this module with 1.0.8, the module cannot communicate with the runtime. After we updating the device sdk nuget packages to 1.20.1, the module seems working properly. There might be some breaking changes from sdk 1.18 to 1.20(or runtime 1.0.5 to 1.0.8)

closed 2022-05-16 by Dolphinsimon