Azure/iot-edge-modbus #83: modbus tcp connection refused on Raspberry pi 3

anfelbeghoura, 2019-12-10:
hey;
I implemented the modbus module on iotedge device  Raspberry pi 3 B, default platform arm32v7 , I put IP address :127.0.0.1 port: 502, I got this logs : 
Connection refused 127.0.0.1:502 
connection lost, reconnecting ...
connect slave failed 
Connection refused 127.0.0.1:502 
connection lost, reconnecting ...
connect slave failed 
.........
first, I tried to make some modifications in the source code to change the port to 1502 but I got the same logs...

Connection refused 127.0.0.1:1502 
connection lost, reconnecting ...
connect slave failed

then, I made some researches, and i found that Ports 0-1024 are "protected" for "known" services. (Check the /etc/services file) and By "protected", they must be used by system daemons. Thus, a regular user cannot start them. so I changed the command line  26 "USER moduleuser" to "USER root" in the docker file but still the same problem ...
I also checked the /etc/services file in the raspberry and I added the Port 502 but always the same issues.
any idea where the problem is?

sandervandevelde, 2020-05-05:
This module is a Modbus 'slave' so there must a Modbus 'master' somewhere.
You refer to the localhost IP address. Is there a Modbus application running on the RPi?
Otherwise, find the right IP address of your Modbus master.

jesweare, 2021-09-09:
It's been quite a while since the last post, but in case others may have seen a similar issue, I figured it would help to chime in:

Assuming that the IP address is set to the `SlaveConnection` desired property field, if the intention behind setting `127.0.0.1` is to poll the host (IoT Edge device) as the Modbus device/slave/server, using `127.0.0.1` may not be the right option. This is because the Modbus module, as an IoT Edge module, is (by default) placed under the Docker daemon's `azure-iot-edge` network.

You can determine the desired IP address by examining the details of the `azure-iot-edge` network bridge and compare that to the network interfaces of your host device. Here is an example in an Ubuntu device I've used for testing.

```
// "172.18.0.1" would be the IP address of the Modbus device
$ sudo docker network inspect azure-iot-edge
[
    {
        "Name": "azure-iot-edge",
        "Id": "f04958f1d3bf175e64d2d4ec696c79afac9572c76012122722ed392c287ca2fc",
        "Created": "2021-07-16T02:42:45.303029015Z",
        "Scope": "local",
        "Driver": "bridge",
        "EnableIPv6": false,
        "IPAM": {
            "Driver": "default",
            "Options": null,
            "Config": [
                {
                    "Subnet": "172.18.0.0/16",
                    "Gateway": "172.18.0.1"
                }
            ]
        },
 <snip>

$ ifconfig
br-f04958f1d3bf: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.18.0.1  netmask 255.255.0.0  broadcast 172.18.255.255
<snip>
```

Hope this helps.

yphuangms, 2022-08-10:
This repository is no longer being actively maintained and will be archived. Open issues will be closed.

closed 2022-08-10 by yphuangms