Azure/iot-edge-modbus #84: Data not being recieved in IoT Central

fetapsomi, 2020-02-23:
I have an edge device connected using the Modbus module.  The module is receiving config, however no data is being sent to IoT Central.  Modbus module logs show that the data is being received from the modbus device but it is not json formatted.  Example below:

440053: 12857

440054: 12848

440055: 12595

440056: 12345

440057: 0

sandervandevelde, 2020-05-05:
The logging is just for.. logging. It does show which registers and coils are read (or written), not the actual JSON outputted. Use a module like https://github.com/iot-edge-foundation/iot-edge-echo/ to inspect the message sent by the output.

yphuangms, 2022-08-10:
IoT Central expect a formatted message like { "field1": "value", "field2": "value" }, hence message sent from modbus modules cannot be correctly consumed by IoT Central. You will need to create a custom module to transfer the message format before routing the message to IoT Central.

yphuangms, 2022-08-10:
This repository is no longer being actively maintained and will be archived. Open issues will be closed.

closed 2022-08-10 by yphuangms