Azure/iot-edge-modbus #87: Only device that contains lots of registers, how to organize data?

rogerbafl, 2020-05-06:
Hello,

I am doing a college project and I'm new on this.

I have to connect and Industrial PLC which work with Windows to Azure, and I consider this solution the best option.

I have to get data from Modbus TCP port 502, where the starting address register I'm interessted in is the 12288 (4012288) and the last is 4014036. Some of the data use one other two registers.

Should I do any modification in the code or can I filter data in Azure?
Do I have to name each register as it is shown in configuration?

I'm pretending to follow this [document](https://docs.microsoft.com/en-us/azure/iot-edge/deploy-modbus-gateway) but I have this questions before starting.

Thanks!

yphuangms, 2020-06-12:
If you want to utilize data ranging in continuous register address space, considering the performance, and the data size, you can read one time from the beginning address with multiple register count as many as you want to, or if the data size is huge, you can separate the read operation into two or tree, if you don't mind the small time difference between reads.

Where you can process the data could be another module, you can route the message to that module to convert the data into meaningful values. In this case, you will need to write an IoT Edge module for this purpose to consume data from your modbus module.

If a cloud side data filtering is preferred, you can also add an Azure Function, or Azure Stream Analytics Job to filter the data from your IoT Hub.  If you prefer not to use any pre-compiled code, you can use Azure Stream Analytics, where you can add an user-defined function written in javascript (https://docs.microsoft.com/en-us/azure/stream-analytics/stream-analytics-javascript-user-defined-functions) to convert the data for next data consumer.

rogerbafl, 2020-06-12:
Hello, 

Thank you for the answer.

Finally I created a function in my Industrial PLC to send data to Azure IoT Hub using MQTT protocol. Then, Stream Analytics is used to filter and organize this data in Azure SQL. 

As you have said, I have small time difference between messages, as per I could not pack all data in the same message, but this has not been any issue in my case.

I will consider this option for any future case as now I am finishing this project.

closed 2020-06-12 by rogerbafl