Azure/iot-edge-modbus #89: Error building image

sihbher, 2020-06-28:
I was trying to add and test some changes but I wasn't able to buil the image, a strange error just show up. I am working on Windows Containers:

Step 11/18 : RUN gcc -shared -o libcomWrapper.so -fPIC comWrapper.c
 ---> Running in 5a7c27a58f60
The command '/bin/sh -c gcc -shared -o libcomWrapper.so -fPIC comWrapper.c' returned a non-zero code: 4294967295: failed to shutdown container: container 5a7c27a58f609c8cd52b25aebf0e04cd390c124fd6437bd467ba4be538aa9ded encountered an error during hcsshim::System::waitBackground: failure in a Windows system call: The virtual machine or container with the specified identifier is not running. (0xc0370110): subsequent terminate failed container 5a7c27a58f609c8cd52b25aebf0e04cd390c124fd6437bd467ba4be538aa9ded encountered an error during hcsshim::System::waitBackground: failure in a Windows system call: The virtual machine or container with the specified identifier is not running. (0xc0370110)

yphuangms, 2020-06-29:
Hi @sihbher, as you mentioned you were working on Windows containers. The native C codes are for Linux containers, because dotnet core 2.x doesn't support serial IO on Linux.
You can only build Linux containers with "Dockerfile.amd64" "Dockerfile.arm32v7" on Linux system.
Please use "Dockerfile.windows-amd64", "Dockerfile.windows-arm32v7" docker files to build Windows containers.

sihbher, 2020-07-02:
thanks for your answer, I would like to add a feature to calculate the float point result of two registers

closed 2020-07-02 by sihbher