Azure/iot-edge-modbus #90: Support for arm64v8

oliverchristen, 2020-09-16:
Is this module still maintained?

We use this module and it works fine. Now we are thinking about using it on an arm64v8 platform which is not supported yet. Does it make sense to create a PR or will it not be merged?

And what about the version 2? Does it get released one time? What are the differences between v1 and v2?

yphuangms, 2020-09-24:
If your arm64 host OS is configured to support arm32, then you can use arm32 container.
To use arm32 container, you will need to hardcode the container as 
`
mcr.microsoft.com/azureiotedge/modbus:1.0.5-arm32v7
`
arm64 Dockerfile and build will definitely be one of our target for next release.
Meanwhile, we encourage you to fork the source to your own repo and customize to fulfill your project need.

Thanks!

oliverchristen, 2020-09-30:
Thank you @yphuangms for your information. We will either try to use the arm32v7 image or build the module by our own. Both sounds feasable.

What about my question above? 
> And what about the version 2? Does it get released one time? What are the differences between v1 and v2?

yphuangms, 2022-08-10:
This repository is no longer being actively maintained and will be archived. Open issues will be closed.

closed 2022-08-10 by yphuangms