Azure/iot-edge-modbus #91: Add content properties to the message sent to the output

sandervandevelde, 2020-10-16:
This is an issue related to this [MS Learn](https://docs.microsoft.com/en-us/answers/questions/128447/how-to-set-the-contenttype-toapplicationjsonand-co.html) question.

Please add these two lines to every outgoing message:

```
message.ContentType = "application/json";
message.ContentEncoding = "utf-8";
```

Aslo, [here](https://github.com/Azure/iot-edge-modbus/blob/master/iotedgeModbus/modules/iotedgeModbus/Program.cs#L324) and [here](https://github.com/Azure/iot-edge-modbus/blob/master/iotedgeModbus/modules/iotedgeModbus/Program.cs#L341), the message is encoded using ASCII encoding.

ASCII encoding is not supported by the IoT Hub routing.  

Is this the right encoding to be used when we add UTF8 encoding later on? What is the risk moving over to UTF8 on these two lines?

qamaruddin, 2020-10-17:
We are also facing the similar issues. The messages stored to blob storage through message routing mechanism are all encoded because of content type.

sandervandevelde, 2020-10-18:
I forked this repo to https://github.com/sandervandevelde/iot-edge-modbus and fixed a few things.

Try my module and enjoy the .net core 3.1 version of iot-edge-modbus.

Some considerations and limitations:

* Prebuilt modules available at https://hub.docker.com/repository/docker/svelde/iot-edge-modbus
* Bumped to .Net core version to 3.1
* Messages now encoded to utf-8 instead of ASCII 
* Added Message properties for ensuring UTF8 encoding
* SQLiteCommandMessage removed
* Deployment limited to Windows/Linux/Raspberry Pi 32bits linux; Non Debug versions
* See https://github.com/iot-edge-foundation/iot-edge-serial#configure-create-options for the Windows Create options

Note: This is tested using the Linux version of the containers only.

yphuangms, 2020-10-20:
Thanks for the information sharing. 

Just wanna clarify that this "ASCII encoding is not supported by the IoT Hub routing." statement is true **ONLY when** utilizing Azure Storage as a routing point.
Please read the document here:
https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-devguide-messages-d2c#azure-storage-as-a-routing-endpoint

**For those who don't setup Azure storage in routing endpoint, the d2c message in ASCII won't cause issue.**

We'll include this fix. And thanks for the catch!

yphuangms, 2022-08-10:
Please fork or clone the code for further modification/customization to build modbus module that fullfill your need.
This repository is no longer being actively maintained and will be archived. Open issues will be closed.

closed 2022-08-10 by yphuangms