Azure/iot-edge-modbus #93: REAL type value is not correct in the output

jasonalang, 2020-10-27:
Dear all,

It seems the REAL type value is not correct in the output as per our test. Did you also encounter the same issue before?

Thank you very much for looking into this, look forward to your insights and advices!

We put the following static values in PLC

PLC Address   Data type    Value
44001             INT              4000
44003             REAL            4002.5
44005             DINT            4004

then we got the following outputs from the Modbus module, so in this case the actual REAL value in PLC is 4002.5, but the output value become 10240

        {
          "CorrelationId": "DefaultCorrelationId",
          "SourceTimestamp": "2020-10-27 06:36:18",
          "Values": [
            {
              "DisplayName": "Test INT",
              "Address": "44001",
              "Value": "4000"
            }
          ]
        },
        {
          "CorrelationId": "DefaultCorrelationId",
          "SourceTimestamp": "2020-10-27 06:35:43",
          "Values": [
            {
              "DisplayName": "Test REAL",
              "Address": "44003",
              "Value": "10240"
            }
          ]
        },
        {
          "CorrelationId": "DefaultCorrelationId",
          "SourceTimestamp": "2020-10-27 06:36:03",
          "Values": [
            {
              "DisplayName": "Test DINT",
              "Address": "44005",
              "Value": "4004"
            }
          ]
        },

yphuangms, 2020-10-28:
Hi, 44003 read from holding register, which is a 2-bytes value. This modbus module implementation will print out this 2-bytes register value as integer value. You will need to convert this integer value back to two bytes and then convert to expected data format.

jasonalang, 2020-10-29:
> Hi, 44003 read from holding register, which is a 2-bytes value. This modbus module implementation will print out this 2-bytes register value as integer value. You will need to convert this integer value back to two bytes and then convert to expected data format.

Thanks very much for your explanation, could you please shed some lights on how to achieve this conversion you mentioned? After converting 10240 to bytes, it's a byte array of 0,40,0,0,0,0,0,0

yphuangms, 2022-08-10:
Please fork or clone the code for further modification/customization to build modbus module that fullfill your need.
This repository is no longer being actively maintained and will be archived. Open issues will be closed.

closed 2022-08-10 by yphuangms