Azure/iot-edge-modbus #95: The While loop in the Start method freezes, when the module has more than 500 active operations

bravenewworlds, 2021-03-19:
I'm trying to read a data from about 100 slaves. Each slave has 50 operations.
In each operation PollingInterval = 1800000 (30min)
PublishInterval = 120000 (2min)
The ping to the slaves is about 50-600ms. Controllers drop connection after 3sec of inactivity. 

When number of slaves > 10, "While" loop in the "Start" method, which must send out messages to IoT Hub,
begins to hang. The code inside the While loop isn't excuted in publish intervals.
The more slaves I'm trying to read, the less often this code is executed.

How can I modify module code to be shure that the code in that loop executes exactly on time (after PublishInterval)?

yphuangms, 2022-08-10:
Usually publish interval should be longer than polling interval, if not, then the content to publish will be repeated with the same polled content. 
The performance of the module can be affected by both the code efficiency and hardware capability.

yphuangms, 2022-08-10:
Please fork or clone the code for further modification/customization to build modbus module that fullfill your need.
This repository is no longer being actively maintained and will be archived. Open issues will be closed.

closed 2022-08-10 by yphuangms