Azure/iot-edge-modbus #96: Cannot read data from slave

arminghasemazar, 2021-08-15:
Hi. 

I'm using Azure portal to deploy modbus module to an IoT edge. Docker logs throws an error as follows

```
Unhandled Exception: System.AggregateException: One or more errors occurred. (Operation timeout expired.) ---> System.TimeoutException: Operation timeout expired.
   at Microsoft.Azure.Devices.Client.InternalClient.<>c.<ApplyTimeout>b__62_2(Task t)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot)
--- End of stack trace from previous location where exception was thrown ---
   at Modbus.Containers.Program.InitEdgeModule() in /app/Program.cs:line 78
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at Modbus.Containers.Program.Main(String[] args) in /app/Program.cs:line 38
```

IoT edge has been installed on Ubuntu 18.04 including all latest packages, IoT edge runtime, etc. I'm using **mcr.microsoft.com/azureiotedge/modbus:1.0.5-amd64**. Below is the Module Twin Settings:

```
{
    "PublishInterval": 2000,
    "SlaveConfigs": {
        "Slave01": {
            "SlaveConnection": "192.168.0.170",
            "HwId": "PowerMeter-0a:01:01:01:01:01",
            "Operations": {
                "Op01": {
                    "PollingInterval": "1000",
                    "UnitId": "1",
                    "StartAddress": "0001",
                    "Count": "16",
                    "DisplayName": "Voltage"
                }
            }
        }
    }
}
```

Network-wise, I can ping the slave device from master (Ubuntu) successfully. I also used [Modpoll Modbus Master Simulator](https://www.modbusdriver.com/modpoll.html) to test whether I can read data from slave and it is working as expected. I also added "Version":"2" to the twin configuration but that also didn't work out.

When I deploy the module, I can see that module starts running for ~2 minutes and then it fails. Any clue where the problem is?

yphuangms, 2022-08-10:
The error message indicates module cannot connect to IoT Edge runtime (could be connection timeout). It could be networking issue between containers and host, or could be IoT Edge configuration issue.

yphuangms, 2022-08-10:
Please fork or clone the code for further modification/customization to build modbus module that fullfill your need.
This repository is no longer being actively maintained and will be archived. Open issues will be closed.

closed 2022-08-10 by yphuangms