Azure/iot-edge-modbus #99: Module healthy but no data received from the device

AsmaAP, 2022-03-09:
I am connecting a "Janitza UMG96RM-E" device to a Windows Laptop where I am using Hyper-V to use Linux cmds. I registered the Laptop as an Edge device and added Modbus to the set of modules. I read the Janitza documentation (start Address, nbre of bits ...) and I Added the correct Ip Address the Module Twins file... Everything looks heathy except I am not receiving any data from the Modbus module... I am not able to figure out what is wrong ? Here is some logs from edgeHub module. 
----------------------------------------------------------------------------------------------------------------------------------------
<6> 2022-03-08 18:32:01.503 +00:00 [INF] - Add node: myEdgeDevice/DefenderIotMicroAgent
<6> 2022-03-08 18:32:01.711 +00:00 [INF] - Add node: myEdgeDevice/Modbus
<6> 2022-03-08 18:32:01.868 +00:00 [INF] - Done refreshing device scope identities cache. Waiting for 60 minutes.
<6> 2022-03-08 18:32:01.956 +00:00 [INF] - Obtained edge hub config from module twin
<6> 2022-03-08 18:32:02.377 +00:00 [INF] - Set the following 1 route(s) in edge hub
<6> 2022-03-08 18:32:02.377 +00:00 [INF] - modbusToIoTHub: FROM /messages/modules/Modbus/outputs/modbusOutput INTO $upstream
<6> 2022-03-08 18:32:02.378 +00:00 [INF] - Updated message store TTL to 7200 seconds
<6> 2022-03-08 18:32:02.379 +00:00 [INF] - Updated the edge hub store and forward configuration
<6> 2022-03-08 18:32:02.391 +00:00 [INF] - Started operation Get EdgeHub config
<6> 2022-03-08 18:32:02.395 +00:00 [INF] - Initialized edge hub configuration
<6> 2022-03-08 18:32:02.397 +00:00 [INF] - Starting protocol heads - (MQTT, AMQP, HTTP)
<6> 2022-03-08 18:32:02.410 +00:00 [INF] - Starting MQTT head
<6> 2022-03-08 18:32:02.524 +00:00 [INF] - Initializing TLS endpoint on port 8883 for MQTT head.
<6> 2022-03-08 18:32:02.845 +00:00 [INF] - Starting AMQP head
<6> 2022-03-08 18:32:02.873 +00:00 [INF] - Started MQTT head
<6> 2022-03-08 18:32:03.251 +00:00 [INF] - Started AMQP head
<6> 2022-03-08 18:32:03.255 +00:00 [INF] - Starting HTTP head
<4> 2022-03-08 18:32:04.026 +00:00 [WRN] - Overriding address(es) '"http://+:80"'. Binding to endpoints defined in "UseKestrel()" instead.
<6> 2022-03-08 18:32:04.063 +00:00 [INF] - Started HTTP head
<6> 2022-03-08 18:32:05.529 +00:00 [INF] - New token received on the Cbs link
<6> 2022-03-08 18:32:05.727 +00:00 [INF] - Client myEdgeDevice/Modbus in device scope authenticated locally.
<6> 2022-03-08 18:32:05.745 +00:00 [INF] - New device connection for device myEdgeDevice/Modbus
<6> 2022-03-08 18:32:05.746 +00:00 [INF] - Client myEdgeDevice/Modbus connected to edgeHub, processing existing subscriptions.
<6> 2022-03-08 18:32:05.750 +00:00 [INF] - Attempting to connect to IoT Hub for client myEdgeDevice/Modbus via AMQP...
<6> 2022-03-08 18:32:05.767 +00:00 [INF] - Bind device proxy for device myEdgeDevice/Modbus
<6> 2022-03-08 18:32:05.767 +00:00 [INF] - Initialized device listener in the AMQP protocol head for myEdgeDevice/Modbus
<6> 2022-03-08 18:32:05.786 +00:00 [INF] - Opened link Events for myEdgeDevice/Modbus
<6> 2022-03-08 18:32:05.827 +00:00 [INF] - Cloud connection for myEdgeDevice/Modbus is True
<6> 2022-03-08 18:32:05.827 +00:00 [INF] - Connection status for myEdgeDevice/Modbus changed to ConnectionEstablished
<6> 2022-03-08 18:32:05.827 +00:00 [INF] - Client myEdgeDevice/Modbus connected to cloud, processing existing subscriptions.
<6> 2022-03-08 18:32:05.827 +00:00 [INF] - Skipping myEdgeDevice/Modbus for subscription processing, as it is currently being processed.
<6> 2022-03-08 18:32:05.827 +00:00 [INF] - Created cloud proxy for client myEdgeDevice/Modbus via AMQP, with client operation timeout 20 seconds.
<6> 2022-03-08 18:32:05.828 +00:00 [INF] - Initialized cloud proxy 9450b673-5293-4c0b-830f-5c564f317321 for myEdgeDevice/Modbus
<6> 2022-03-08 18:32:05.828 +00:00 [INF] - Created cloud connection for client myEdgeDevice/Modbus
<6> 2022-03-08 18:32:06.116 +00:00 [INF] - Opened link TwinReceiving for myEdgeDevice/Modbus
<6> 2022-03-08 18:32:06.126 +00:00 [INF] - Processing pending subscriptions for myEdgeDevice/Modbus
<6> 2022-03-08 18:32:06.129 +00:00 [INF] - Opened link TwinSending for myEdgeDevice/Modbus
<6> 2022-03-08 18:32:06.659 +00:00 [INF] - Processing pending subscriptions for myEdgeDevice/Modbus
<6> 2022-03-08 18:32:06.661 +00:00 [INF] - Opened link ModuleMessages for myEdgeDevice/Modbus
<6> 2022-03-08 18:32:06.760 +00:00 [INF] - Updated reported properties for myEdgeDevice/$edgeHub
<6> 2022-03-08 18:32:12.133 +00:00 [INF] - Updated reported properties for myEdgeDevice/Modbus
<6> 2022-03-08 18:36:56.469 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 18:41:56.461 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 18:46:56.461 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 18:51:56.464 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 18:56:56.464 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 19:01:53.016 +00:00 [INF] - Started task to cleanup processed and stale messages for endpoint iothub
<6> 2022-03-08 19:01:53.083 +00:00 [INF] - Cleaned up 0 messages from queue for endpoint iothub and 0 messages from message store.
<6> 2022-03-08 19:01:56.460 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 19:06:56.460 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 19:11:56.463 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 19:16:56.463 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 19:21:56.463 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 19:26:56.464 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 19:30:05.794 +00:00 [INF] - New token received on the Cbs link
<6> 2022-03-08 19:30:05.800 +00:00 [INF] - Token updated for myEdgeDevice/Modbus
<6> 2022-03-08 19:30:57.126 +00:00 [INF] - Request starting HTTP/1.1 GET http://172.18.0.3:9600/metrics
<6> 2022-03-08 19:30:57.203 +00:00 [INF] - Request finished in 76.276ms 200 text/plain; version=0.0.4; charset=utf-8
<6> 2022-03-08 19:31:53.089 +00:00 [INF] - Started task to cleanup processed and stale messages for endpoint iothub
<6> 2022-03-08 19:31:53.157 +00:00 [INF] - Cleaned up 0 messages from queue for endpoint iothub and 0 messages from message store.
<6> 2022-03-08 19:31:56.459 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 19:32:01.877 +00:00 [INF] - Starting refresh of device scope identities cache
<6> 2022-03-08 19:32:02.390 +00:00 [INF] - Starting periodic operation Get EdgeHub config...
<6> 2022-03-08 19:32:02.781 +00:00 [INF] - Obtained edge hub config from module twin
<6> 2022-03-08 19:32:02.939 +00:00 [INF] - Set the following 1 route(s) in edge hub
<6> 2022-03-08 19:32:02.940 +00:00 [INF] - modbusToIoTHub: FROM /messages/modules/Modbus/outputs/modbusOutput INTO $upstream
<6> 2022-03-08 19:32:02.941 +00:00 [INF] - Updated message store TTL to 7200 seconds
<6> 2022-03-08 19:32:02.941 +00:00 [INF] - Updated the edge hub store and forward configuration
<6> 2022-03-08 19:32:02.941 +00:00 [INF] - Successfully completed periodic operation Get EdgeHub config
<6> 2022-03-08 19:32:03.056 +00:00 [INF] - Updated reported properties for myEdgeDevice/$edgeHub
<6> 2022-03-08 19:32:04.404 +00:00 [INF] - Not changed node: myEdgeDevice
<6> 2022-03-08 19:32:04.405 +00:00 [INF] - Not changed node: myEdgeDevice/$edgeAgent
<6> 2022-03-08 19:32:04.405 +00:00 [INF] - Not changed node: myEdgeDevice/$edgeHub
<6> 2022-03-08 19:32:04.405 +00:00 [INF] - Not changed node: myEdgeDevice/DefenderIotMicroAgent
<6> 2022-03-08 19:32:04.406 +00:00 [INF] - Not changed node: myEdgeDevice/Modbus
<6> 2022-03-08 19:32:04.406 +00:00 [INF] - Done refreshing device scope identities cache. Waiting for 60 minutes.
<6> 2022-03-08 19:36:56.459 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 19:41:56.461 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 19:46:56.462 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 19:51:56.460 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
[... cut ...]
<6> 2022-03-08 20:01:56.463 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 20:06:56.461 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 20:11:56.460 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 20:16:56.463 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2022-03-08 20:21:56.461 +00:00 [INF] - Entering periodic task to reauthenticate connected clients

yphuangms, 2022-08-10:
You can check the log of modbus module (docker logs <modbus_container_name>) to get more details about modbus module runtime condition/status.

yphuangms, 2022-08-10:
Please fork or clone the code for further modification/customization to build modbus module that fullfill your need.
This repository is no longer being actively maintained and will be archived. Open issues will be closed.

closed 2022-08-10 by yphuangms