Azure/iot-edge-modbus #100: No longer being maintained

mahebert, 2022-05-14:
We are using this module for a new product that we develop. I saw that this repository is archive and no longer maintained.

So what are the alternatives, if we want to read data from multiple PLC (form multiple vendors) ?

rploeg, 2022-05-14:
You find the latest module here: https://github.com/Azure/Industrial-IoT

yphuangms, 2022-08-10:
Current implementation of modbus module can support connection to multiple PLC (by setting multiple SlaveConnection). Please check the documentation for more details.

yphuangms, 2022-08-10:
Please fork or clone the code for further modification/customization to build modbus module that fullfill your need.
This repository is no longer being actively maintained and will be archived. Open issues will be closed.

closed 2022-08-10 by yphuangms